hzeller/rpi-rgb-led-matrix #13: Scrolling ppm

bart178, 2015-01-13:
I am using your project to control a 16x32 led matrix

Particular the demo1 scrolling message
As you can see in the adafruit tutorial i'm missing THE lowest horizontal row and first /last vertical  row on the right
https://learn.adafruit.com/connecting-a-16x32-rgb-led-matrix-panel-to-a-raspberry-pi/overview

Any idea how i can solve this?

I'm using a  rpi and led matrix to scrol ppm's generated by my restored arcade cabinet's frontend
80's Pacman style ! Gives my project a personal touch!

Thanx for sharing your code!

hzeller, 2015-01-13:
On 13 January 2015 at 13:41, bart178 <notifications@github.com> wrote:

> I am using your project to control a 16x32 led matrix
>
> Particular the demo1 scrolling message
> As you can see in the adafruit tutorial i'm missing THE lowest horizontal
> row and first /last vertical [...]

I heard that before from people using the Adafruit tutorial.
Do you actually use my code on
https://github.com/hzeller/rpi-rgb-led-matrix or the code on Adafruit ?
Adafruit forked my code some time ago, so it is kindof ancient right now -
it might contain a bug.

-h

bart178, 2015-01-14:
Thanx for your reply!
Now used your code, 

Now image is upside down (no problem, i can turn matrix) 
But  if i scrol a 16pixel heigh ppm      only 8 rows are used   scrolling the 16 rows  (seems that all pixels squeezed over the first 8 rows ...

if i scrol a 32 pixel ppm     al 16 rows are used (still a bit blurry)

Any idea what's wrong?  

tought about a wiring issue ?   

Greetings

hzeller, 2015-01-14:
>
> Now image is upside down (no problem, i can turn matrix)
>
> But if i scrol a 16pixel heigh ppm only 8 rows are used scrolling the 16
> rows (seems that all pixels squeezed over the first 8 rows ...
>

did you use the -r16 commandline flag ? By default, the code assumes a 32
pixel high display; to use a 16 row display, you need to give it -r 16.

-h

hzeller, 2015-07-01:
Looks like things worked fine ? Closing now this issue. If you have further problems, please re-open.

closed 2015-07-01 by hzeller