hzeller/rpi-rgb-led-matrix #14: Raspberry Pi 2

helgevh, 2015-02-14:
Hi,
I have successfully used your program on my Raspberry Pi Model B. 
Now I have tried to use it on my new Raspberry Pi 2 Model B. I am able to compile the code but the RGB Matrix is staying black when I run the program. I guess the new hardware of the Rapsberry Pi 2 needs some code fixing. Do you have some hints, where to start?

Regards 
Helge

hzeller, 2015-02-14:
On 14 February 2015 at 08:14, helgevh <notifications@github.com> wrote:

> Hi,
> I have successfully used your program on my Raspberry Pi Model B.
> Now I have tried to use it on my new Raspberry Pi 2 Model B. I am able to
> compile the code but the RGB Matrix is staying black [...]

I don't have a Raspberry Pi 2 yet, so I don't know what is wrong; as soon
as mine arrives, I'll dig into it. Until then - if you find out something,
let me know :)

-h

hzeller, 2015-02-14:
On 14 February 2015 at 08:36, Henner Zeller <h.zeller@acm.org> wrote:
> On 14 February 2015 at 08:14, helgevh <notifications@github.com> wrote:
>
>
> I don't have a Raspberry Pi 2 yet, so I don't know what is wrong; as soon as
> mine arrives, I'll dig into it. Until then - if [...]

Looks like adafruit cloned my repository and they did something to
make it work with the Pi 2
https://github.com/adafruit/rpi-rgb-led-matrix/commits/master

-h

>
> -h
>

helgevh, 2015-02-14:
Thanks for the hint. I've tested the adafruit version but I am not able to build it. I think tomorrow I will have a closer look at it.

Helge

perl5577, 2015-02-15:
need python-2.7-dev and other package for image
Not work for me :(

helgevh, 2015-02-15:
Same here. Matrix is staying black.

hzeller, 2015-02-15:
On 15 February 2015 at 10:25, helgevh <notifications@github.com> wrote:

> Same here. Matrix is staying black.
>
Interesting. I'll try to get a new Raspberry Pi to see...

perl5577, 2015-02-16:
work with HAT adafruit :(
https://www.youtube.com/watch?v=EvcBqRSVU7U

probleme converter 3.3V > 5V ?

hzeller, 2015-02-16:
On 15 February 2015 at 16:01, perl5577 <notifications@github.com> wrote:

> work with HAT adafruit :(
> https://www.youtube.com/watch?v=EvcBqRSVU7U
>
> probleme converter 3.3V > 5V ?
>

no, the RGB panels work with 3.3V signal level; the 5V converter is just
there to give an extra margin.

perl5577, 2015-02-16:
why work with HAT adafruit ? 
Only difference between direct connexion and HAT is converter level .

hzeller, 2015-02-16:
>
> why work with HAT adafruit ?
> Only difference between direct connexion and HAT is converter level .
>

Well, don't ask me, it is Adafruits product :)
I guess it makes it simpler to get things running right away with an easy
plug-and-play board.

They might have changed as well some GPIO pins to work with their hat, so
you might want to look out for that as well when you use that code.

hzeller, 2015-02-16:
RPi 2 is on the way, I'll look at it as soon as I get it.

helgevh, 2015-02-16:
They have added a "#ifdef ADAFRUIT_RGBMATRIX_HAT" section in several files, so you can select between Adafruit hat and the old pinout. The changes aren't mentioned in the readme file.

changed files
lib directrory:
framebuffer.cc
framebuffer-internal.h
Makefile
root directory:
Makefile
makefile

perl5577, 2015-02-16:
More simple 
in ./lib/Makefile  comment line DEFINES+=-DADAFRUIT_RGBMATRIX_HAT
in ./Makefile remove -DADAFRUIT_RGBMATRIX_HAT  for cible .o

helgevh, 2015-02-16:
Is it working for you? I have already removed both entries and recompiled the files. Now I see some flickering on the RGB matrix. But the test images are still not working.

perl5577, 2015-02-16:
is it  working for me,  I use both  32x64 . ( -L )
I have reduce flickering  , I load 100% one core .

http://monero.crypto-pool.fr/videos/IMG_1146.MOV
http://monero.crypto-pool.fr/videos/IMG_1148.MOV

helgevh, 2015-02-17:
Nice work. Especially the flames are looking great. Are you using pixel-pusher? https://github.com/hzeller/rpi-matrix-pixelpusher

closed 2015-02-17 by helgevh