hzeller/rpi-rgb-led-matrix #16: Issues driving 32x64 matrix

rogerguess, 2015-02-18:
I have a 32x64 matrix from Adafruit, and have been unable to drive a consistent pattern across all rows. I had at first suspected that the matrix had issues, but some of the demos seem to suggest all pixels are capable of all desired colors. Note that at 1:15 in the video, I cycle through demo 0 with each bit value 1 to 11 for PWM.

http://youtu.be/_QmmxjrfeGI

I am running the led-matrix with flags for rows at 32 and chained at 2 (yes, when chained = 1, it acts like 2 32x32 panels, but they are not perfectly mirrored). I made sure all GND pins on the panel are connected to the Pi and I am using a 10A power supply.

hzeller, 2015-02-20:
Mmh, it looks a bit like it could be that the logic voltage levels are not properly registered, and maybe the panel puts too much load on the pins. I'd try to use some HCT245 as 3.3V -> 5V level shifter.

(Looks like Adafruit made a HAT that has level shifters built-in https://www.adafruit.com/products/2345 )

Power supply sounds good, just make sure to have good beefy cables connected towards the panel (your dual-panel easily uses 6A with very short even higher current bursts). A largish capacitor directly at the panel is probably good as well (see README).

rogerguess, 2015-02-20:
Yes, ordered a HAT that arrives next week. Will test then and share what I find. Will read up on the capacitor. Thank you.

hzeller, 2015-02-27:
I found that some panels have trouble with the signal speed on the newer RPi2. I have now slowed down writing to it and found that it actually helped for one on my problematic panels.

Can you do a fresh pull and try your 32x64 panel ?

makifbuz, 2015-02-27:
I am away from my home. I can try it on sunday.  Frankly as you know i
don't have problem with 32x64 but i made 32x160 display and the problem on
the 9.th and 10.th panel. But if you slowed down the sending data it would
work without increasing the power supply voltage. And i saw that distance
between first panel and the rpi2 is important for the clearance of the data.

rogerguess, 2015-02-27:
I have the HAT from adafruit in place. I pulled latest here, but nothing is showing on the display at all. I also tried the latest from adafruit's fork and all is working fine. I'll look into what they have done to see if there is something specific I need to do for the HAT. Hopefully just a pin mapping.

hzeller, 2015-02-27:
yes, they did some pin-mapping. They also have done some things like the RPi2 detection that is a bit different from mine (they forked from a version before I had a RPi2, so I didn't have that at yet at that point). Mostly you need to look out for the pin-mapping around here
  https://github.com/adafruit/rpi-rgb-led-matrix/blob/master/lib/framebuffer-internal.h#L67

If you pull latest here and don't use the HAT, I would be interested if it now works even without the hat as I slowed down the clocking a bit.

rogerguess, 2015-02-27:
I currently have no other means for level shifting other than the HAT. I need to learn about those. I am afraid the same issues I had with only supplying 3.5 would hide any improvements you created.

I love the HAT, btw. I am even powering the Pi with it: https://forums.adafruit.com/viewtopic.php?f=47&t=68718

rogerguess, 2015-02-28:
Sorry, not having any luck sorting out what all is needed to get the HAT working with your repo. I love the HAT, but hope adafruit's fork calls back home to yours soon. Can you share basic advice on where/what to get for level shifting?

hzeller, 2015-03-04:
For level shifting, I'd just use some HC245 bus driver.
I prepared a couple of PCBs in https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter - look for the 'active' ones. I have not tried these yet (still waiting for the order to arrive), but they should work.

rogerguess, 2015-03-04:
I tried without the HAT (so I have no level shifting) and it is very close. I seem some 'dancing' of pixels on demo 3, for example. Also some inconsistency at different alpha levels with demo 5 (top left is barely lit). http://youtu.be/hILPU1NHjt8  (still uploading)

hzeller, 2015-03-05:
So I now switched off the default slow-down of the GPIO I had before; it was hurting in particular older Raspberry Pis, that had redueced frame-rate because of that.

However, I described things here:
   https://github.com/hzeller/rpi-rgb-led-matrix#help-some-pixels-are-not-displayed-properly

So with this, your original direct connection without the HAT of the 32x64 matrix will be erratic again, but you can get this fixed by uncommenting the DEFINE as described in the README.

I'd be interested if it actually properly works with the HAT without the slowdown trick (now that you have a branch of my repository that works with the HAT as well). Because that would mean that the level shifting is actually helping.

hzeller, 2015-07-01:
Haven't heard anything in this thread, but I have seen the slow-down option helping in other cases.
Marking this now as closed. If you see further issues, please re-open.

closed 2015-07-01 by hzeller