hzeller/rpi-rgb-led-matrix #17: Docs: compiler optimization flag

CaraesNaur, 2015-02-19:
Please add to the documentation that images will not be displayed with strictly correct colors unless -O3 is passed to the compiler.  The files get loaded and parsed correctly, but without the optimizations the code doesn't run fast enough and the output appears "posterized", however this is a side effect of the timing.

hzeller, 2015-02-19:
Ok, but isn't that already in the Makefile ?
If someone messes with that, I would presume they already know what they are doing. But yeah, I'll add that to the README.

CaraesNaur, 2015-02-19:
It is in the makefile here, however we were using this as a library and wrote our own makefile and somehow ended up omitting that flag.  Much furious and fruitless debugging was done until we realized that.

commit 10bde55a (2015-02-22): o Update doc about optimization. Fixes #17
closed 2015-02-22 by hzeller