hzeller/rpi-rgb-led-matrix #18: Pi 2 Model B

rogerguess, 2015-02-20:
My Pi 2 arrived. I have tried the connection instructions from the readme, and I used http://elinux.org/RPi_Low-level_peripherals as a map to the 40pin header (using the section on Model A+ and B+). I am getting nothing at all on a 32x64 matrix.

Any suggestions?

hzeller, 2015-02-20:
My PI 2 just arrived as well. I'll check it out on the weekend.

rogerguess, 2015-02-20:
Excellent, look forward to what you find out.

FYI, I found this:
http://raspberrypi.stackexchange.com/questions/27268/do-the-gpio-registers-have-the-same-addresses-on-the-new-raspberry-pi-2/27270#27270  suggesting a new base address, but I was unsuccessfully trying those with gpio.cc

rogerguess, 2015-02-20:
Ah, and this: https://github.com/adafruit/rpi-rgb-led-matrix/commit/7cea167e4d2b63250174a79c9420478f7edc3c2c

hzeller, 2015-02-20:
Yes, looks like they worked around that. yay.
I now have three versions: RPi 1, RPi 1 B+ and a new RPi 2 B that I'll test on the weekend and update the code.

hzeller, 2015-02-22:
Fixed in f2b5e0cd56adce913dd5f3e8d588f8f10c0daf47
Code works now as well with RPi 2 (there might be some refinements needed regarding timing, but the basics work)

closed 2015-02-22 by hzeller

rogerguess, 2015-03-04:
Confirmed, works well. Thank you.