hzeller/rpi-rgb-led-matrix #21: Running when Pi boots up.

rogerguess, 2015-03-10:
I am attempting to have the led-image-viewer run when the Pi boots up.

I have followed the instructions here:
http://www.raspberry-projects.com/pi/pi-operating-systems/raspbian/auto-running-programs

Is there anything that would prevent the led-image-viewer to run if no ssh session is open on the Pi?

mutax, 2015-06-28:
Please state exactly what you have done, as it is hard to guess where the problem is.
How did you call the image viewer and with what parameters?
A ssh connection will not be needed for the programs to run.

hzeller, 2015-07-01:
One important thing you might need is the `-d` option that all the binaries provide; it makes it a daemon, which disconnects from any terminal.

closed 2015-08-07 by hzeller