hzeller/rpi-rgb-led-matrix #26: LEDs flicker

bmoore27, 2015-04-18:
I have an Adafruit 64x32 LED Matrix (4mm pitch), a Adafruit RGB Matrix HAT (https://www.adafruit.com/product/2345), a Raspberry Pi B+ and a 5v 4W power supply.

I have everything wired up correctly (i think) the matrix responds and I am able to display text and images. However, my issue is that rows of LEDs flicker, almost like a bad refresh rate. You can see the video of the issue here: http://tinyurl.com/pt67son and attached.

What causes the flicker? Is it that the rpi can't refresh the display fast enough?

hzeller, 2015-04-18:
This is a known issue (see 'Limitations' in README). Sorry about that.

The problem is, that the Pi is not running a realtime operating system and thus the duration of LED on varies a bit, which shows as these glitches.

You can reduce this by using less PWM width. If you set your text to 'full on' , you can set the pwm width to 1 (-p 1). This reduces the number of total colors displayed to 8, but you will see much less flicker (if at all).

Also it helps to have nothing else running on the system. So for instance no graphical user interface or anything, just have a standalone Raspberry Pi where you log in via ssh.

I do have a plan to reduce this by doing some hardware tricks (using DMA instead of doing it by the processor), but probably not before 2 months from now (other projects going on right now).

BennoG, 2015-05-25:
I am testing the GPIO18 PWM output mode on the ENABLE input of the matrix.
wit the example code from http://www.raspberrypi.org/forums/viewtopic.php?f=72&t=67741
The problem is that it will only work on GPIO18 but it seems to work stable

interval1066, 2015-05-25:
>The problem is, that the Pi is not running a realtime operating system...

Henner;

You're making (a probably safe) assumption that Beau is running a
non-deterministic OS, so are you saying the issue would deffinately be
resolved if the RasPI was running, say, AngstromOS or summat?

hzeller, 2015-05-25:
@BennoG this sounds good - if the PWM can be programmed arbitrarily to generate a particular pulse length, then the flicker could be fixed. Thanks, I'll try that avenue next weekend.
(and probably have to find some nightmarish way to make sure people can be safely converted to the 'new' pinout).

BennoG, 2015-05-25:
After some testing today with my own project to use HUB8 HUB12 and HUB75 witch is using your GPIO class.
I have changed the GPIO class to use the PWM pulse (negative enable and positive enable),
Had some trouble to get the pulse stable while filling and waiting for the fifo.
seems to be solved right now, I will cleanup the code and compare to your GPIO and try if i can submit the changes. Then you can generate pulse with nanosecond precision ;)

BennoG, 2015-05-25:
I have cleaned up the PWM code and isoated it to a seperate class (because it had little to do with the GPIO class besides PI2 detection and memory mapping)

http://www.ans-net.nl/dl/pwmPulse.zip

usage:
pwmPulse p;
p->sendPulseNano(100);

hzeller, 2015-05-28:
Thanks @BennoG, I'll try that next spare minute on the weekend.

hzeller, 2015-07-01:
BTW, an update on this: the suggestion looks promising. I have an experimental branch experimental-pwm-bit in which this is implemented (and yeah, it will require some change in the wiring, so I need to update the documentation etc. first before that is merged with main).

The flicker is gone with that (yay), however I need a little bit more CPU, which I'll try to fix first before merging with main.

Saij, 2015-07-01:
YAY
Wanted to try out the branch when I'm done building my mobile LED Matrix playground suitcase ^^
Good to here that the PWM Pin helps. (CPU is no problem for me on the RPi2)

Saij, 2015-07-01:
Do you think the RT patches for the kernel will work here better?
Just wondering if I should recompile the kernel with the RT patches.

hzeller, 2015-07-03:
I tried the RT patches a while back, and they made a little, but not very impressive difference. In particular with the new PWM based code, it probably doesn't make much of a difference.

Saij, 2015-07-03:
Will give it a test (if I get the kernel compiling and not destroying my RPi with optimizing ^^)

hzeller, 2015-07-03:
See the README.md a the end where I link to a pre-built RT image if you want to try to that first.

I'll hopefully get DMA working over the weekend, so things might improve considerably wrt. lower CPU usage (and the jitter is already pretty optimal with the pwm branch).

Saij, 2015-07-03:
Tried that RT Image. But couldn't login and couldn't find any credentials.
But I'm going to compile the RT Kernel from the same Repo as the kernel in the image (one time I already built it - but I destroyed my Pi trying to optimize boot ^^)

I'm locking forward to the DMA update. PWM branch is currently working very well (showed Demo 4 with 11 PWM Bits to my co-workers and they all told me they can't see any flicker)

hzeller, 2015-07-20:
Update: the PWM changes are now in the master branch. Whoever had trouble with flickering LEDs, this should fix it. Note: **the pinout changed**.

I investigated the use of DMA, and it is not worthwhile, as DMA is too slow ( Gory details here https://github.com/hzeller/rpi-gpio-dma-demo ).

hzeller, 2015-07-25:
Closing this now. The new pinout and hardware support for timing allows for a very smooth and flicker-free experience.

closed 2015-07-25 by hzeller

bmoore27, 2015-09-05:
hzeller, could you point me to the correct pinout layout. I have connectors labeled HUB75 but nothing else is written on the pins.

hzeller, 2015-09-05:
There are two pictures in the README showing the pinout of the Hub75 and a picture of IDC connector how it should look like there.
Typically, the displays have two headers, one of it not labelled - that is the output. If both are not labelled, then look for arrows indicating the direction (its all described in the readme).

bmoore27, 2015-09-05:
I may be dense, I see the pinouts of the Hub75 connector in the images but your note that the pinouts have changed do not indicate to me whats changed. 

I assume my Hub75 pinouts is what is show in the images:

LEFT | RIGHT | NOTCH
---- | ----- | -----
G1 | R1 | 
GND | B1 | NOTCH
G2 | R2 | 
GND | B2 | 
B | A |
D | C | 
STROBE | CLK | NOTCH
GND | OE |

hzeller, 2015-09-07:
Ah, I see. So Hub75 did not change, it is the same as before - only the pinout of the Raspberry Pi changed, so on which GPIO pins the Raspberry Pi has signals.

(If you want to see how it was before, you can look at the readme of this old version
https://github.com/hzeller/rpi-rgb-led-matrix/tree/83b8bf4bc26043a8b62fbd002ed5174fbb8d7a5f
)

Your Hub75 pinout above looks correct, but not that there is only one notch in the middle.

bmoore27, 2015-09-17:
So I think I know why I am confused.

I have a 32x32 + 32x32 panel, an Adafruit RGB Matrix HAT and a Raspberry Pi B+ with a 40pin connector.

Right now the 40pin connection is direct between the HAT and Pi but since the pinouts are different I will need to use jumper wires to change the pin out. The pinouts that have changed will require me to use jumper between the Adafruit HAT and the Pi, right?

jwheare, 2015-11-19:
Hi, I also have an Adafruit HAT with 64x32 panel and running into a lot of glitching with the demo using adafruit's fork of this project here: https://github.com/adafruit/rpi-rgb-led-matrix

I even get the flickering using the -p 1 setting.

Predictably, running with your latest code from this repo doesn't work. Can you please confirm that jumpers will need to be used to convert the pinouts from the Adafruit HAT to work with your latest code?

jwheare, 2015-11-19:
Here's a short video of the flickering https://www.youtube.com/watch?v=kRfAhIhxIFg with the following command line options:

    sudo ./led-matrix -r 32 -c 2 -D 0

hzeller, 2015-11-19:
Please create a bug that is related to your problem, not use some entirely unrelated and fixed bug @jwheare - it is entirely confusing for everybody involved.

It looks like you either need level shifting in your connection or use the GPIO-slowdown. Did you follow the steps described in the following link ?
https://github.com/hzeller/rpi-rgb-led-matrix#help-some-pixels-are-not-displayed-properly

jwheare, 2015-11-19:
Sorry, it appeared to be a related issue to this, my question was just a reiteration of the unanswered question from the original reporter in the comment before with my own specifics added for clarity.

Thanks for the link, I'll take a look. I'm also considering whether it's a faulty panel.

hzeller, 2015-11-19:
The original poster showed brightness glitches in single lines, which was indeed a problem if you can't use PWM. In your case, random pixels show up that shouldn't.

(if you use the Adafruit HAT directly, you can't use the PWM either as well, you need to do the bridging between two GPIOs as described in the lib/Makefile).

The Panel is probably not faulty, you just need to make sure that the logic levels are right.

jwheare, 2015-11-19:
Thanks, the GPIOS slowdown fix solved it for me. That section was missing from the README in adafruit's fork. I'll file an issue with them for that.

hzeller, 2015-11-19:
Yeah, don't use the Adafruit fork - it is ancient.