hzeller/rpi-rgb-led-matrix #27: Multiple Matrix Displays

sorphin, 2015-04-25:
The Demo code can do 64x64 from 4 displays (and I've made it work as 96x64 and 128x96).. However, The image displayer program uses the Frame Canvas. I'm unable to find how to make the LargeSquare64x64Canvas type code work for the Frame Canvas so the image code can do it as well.

sorphin, 2015-05-10:
Nothing?

hzeller, 2015-05-10:
What is exactly the problem, can you elaborate ?
Do you have some code-examples that you tried but failed to work ?

sorphin, 2015-05-11:
Your image Demo program in the library. led-image-viewer.cc   it cannot handle 6 matrices in a 96x64 formation (Whether Z or C wiring).. That code uses the "Frame Canvas" from the library, so the same changes made to LargeSquare64x64Canvas won't work. I tweaked LargeSquare64x64Canvas to LargeSquare96x64Canvas and LargeSquare128x96Canvas and the programs that use those work, because they don't use the 'Frame Canvas" function. What I need to know is how to modify the Frame Canvas code to do the same so I can do large scale images like I can do large scale on the demo functions.

hzeller, 2015-05-11:
Ah. Just wrap a FrameCanvas with your implementation of a LargeScale canvas. The FrameCanvas just implements a Canvas in itself, and your LargeScale canvas that does the pixel-mapping can use that as a backend.
You can just allocate your LargeScale canvas on the stack in the PreprocessedFrame constructor while filling in the FrameCanvas (later on, only these already mapped and preprocessed frames are sent to the output, which means there is no overhead in the mapping). Of course, you need to make sure to _not_ delete the delegatee canvas in the destructor otherwise bad things will happen :)

sorphin, 2015-05-11:
I have a feeling I'm gonna have a headache after this. ;-)

hzeller, 2015-09-13:
In the meantime, there is now the concept of a 'Canvas Transformer'
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/canvas.h#L51

An implementation can handle all the transformations needed for such pixel mapping. Then you can use this as a wrapper to manipulate other canvases. It is not perfect, but it simplifies the task.

For implementations, have a look at 
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/transformer.cc

hzeller, 2015-09-13:
(closing for now, not much activity here. Let me know if you need more)

closed 2015-09-13 by hzeller