hzeller/rpi-rgb-led-matrix #30: Some additional examples

jostien, 2015-06-07:
Hi,

Thanks for the API! I use the LED-matrix in winter for running to work and back home. I fix it to my running rucksack and then it shows several pong matches so that cars don't run me over.  I put some additional examples which I use below:

https://github.com/jostien/rpi-rgb-led-matrix-examples

The code is almost a year old and I hope that it goes along with the current version or rpi-rgb-led-matrix. There is stuff like Beavis and Butthead doing some heavy metal or GIR dancing (GIR is not in the repository because it is a variation of B&B). Unfortunately I cannot provide the pngs because I doubt that I have the copyright.

Pong was derived from xscreensaver. So there is a lot of material which could be adapted to the LED-matrix. For instance Pacman would be nice. However, days are long in the moment, so maybe in autumn ...

Regards,
jost

hzeller, 2015-07-01:
Nice. I am soon going to break out the examples in a separate repository so that it is easier to contribute new examples. So if you are ready in autumn, I am happy to see a pull request :)
(I'll update this thread when I have a repository name for the examples)

Dhanielk, 2017-11-25:
Hi @hzeller. Is there any way to use API to feed it with 2D Array? That will be really helpful for my project.

Thanks,
Daniel

hzeller, 2017-11-25:
just call SetPixel() in a loop. It would not be significantly faster if there was a call with a 2d buffer.

Dhanielk, 2017-11-26:
Ok. Got it... If I have time, I will create PR for it.

combs, 2023-04-23:
I wonder if this issue is still relevant?