hzeller/rpi-rgb-led-matrix #32: led-image-viewer doesn't work. "Crashes"?

Cyb0rk, 2015-06-10:
Hi there,

I compiled led-image-viewer and tried it on several gif images, but all I get is the following output:

pi@autopi ~/display16x32/rpi-rgb-led-matrix $ sudo ./led-image-viewer giphy.gif
Read image...
Assembling animation with 25 frames.
Scale ... 281x216 -> 32x32
Preprocess for display.
Display.

Its runs for about half a second on terminal and than terminates. And nothing is shown on the panel. Rest of the examples work fine!

Any Idea?

hzeller, 2015-06-10:
Can you attach the image that is breaking ?

Cyb0rk, 2015-06-10:
Of course, here are two of them.

[image]
[image]

artictriac, 2015-06-13:
I'm having this same issue. Tried all sorts of image formats, sizes, shapes, etc. 

I get the below every time just with the scale line changed to match whatever image I'm trying to use.

Read image...
Scale ... 500x32 -> 32x32
Preprocess for display.
Display.

Nothing on the display, not even a flicker. Program exits and I'm left with the command prompt. 

Raspberry Pi 2 B v1.1
Running fresh install of Raspbian
Debian Wheezy
Release date: 2015-05-05
with all dependencies listed in your readme installed. Nothing extra over a default install.
4GB Class 4 SanDisk Micro SD card

./led-matrix runs perfectly with all included demos and any correctly formatted .ppm image

hzeller, 2015-06-13:
Should be fixed now, can you try again ?

closed 2015-06-13 by hzeller

artictriac, 2015-06-13:
It is now working for me. Thanks for fixing it so quickly.

Blevo90, 2015-06-22:
That fixed my issue but now I cannot get the display to stop without killing the process

hzeller, 2015-06-22:
On 21 June 2015 at 19:08, Blevo90 <notifications@github.com> wrote:

> That fixed my issue but now I cannot get the display to stop without
> killing the process
>

Ctrl-C doesn't work ?

Blevo90, 2015-06-22:
yes it does but I am a newbie to linux and the other program says what to
do this one did not. A friend clued me to the ctrl+c

hzeller, 2015-06-22:
On 21 June 2015 at 20:18, Blevo90 <notifications@github.com> wrote:

> yes it does but I am a newbie to linux and the other program says what to
> do this one did not. A friend clued me to the ctrl+c

Yes, they are inconsistent. The other demo programs don't get shut down
properly on Ctrl-C, they do return.
I'll make these conistent soon.

commit d2a89aca (2015-06-13): o Fix #32