hzeller/rpi-rgb-led-matrix #34: Active 3 Boards are no longer available.

squeegiefreak, 2015-06-18:
I went to order the Active 3 boards, and it looks like they are no longer available.

hzeller, 2015-06-18:
On 17 June 2015 at 17:33, Courtney Powers <notifications@github.com> wrote:

> I went to order the Active 3 boards, and it looks like they are no longer
> available.
>

Yes, I am currently working on improving the library, but unfortunately it
means that some pinout change.
This development is taking place in some branch (experimental-pwm-bit)
Since this is soon to change, I removed the boards to not add confusion and
have people set some compatibility flags.

But I have some new boards with the new layout that I haven't gotten back
yet from OshPark, so couldn't test, but I can share them if you like (also,
you need to use the new branch, it it is likely that it is active when the
boards arrive).

I'll update this thread later this evening with the links to the
(tentative, but probably ok) boards.

hzeller, 2015-06-18:
This is the updated OshPark link https://oshpark.com/shared_projects/fIYlJxnA

I have updated the links in the experimental branch
 https://github.com/hzeller/rpi-rgb-led-matrix/tree/experimental-pwm-bit/adapter/active-3
Note, these are not tested yet, and there might be a slight chance that I have to do some little changes to the wiring, so no guarantees at this point :)

Also, when you use this guy, be prepared to run code from this branch (though I think I've finalized that mostly once you receive your order from OshPark).

Note, I changed the package from TSSOP20 to SOIC-20 as this might be easier to hand-solder for the average person.

squeegiefreak, 2015-06-18:
Ok I'll give it a shot unfortunately I just ordered the tssop package from digikey. I'll see if I can change the order to the new one package. Do you have an estimate for the new rev boards arrival?

hzeller, 2015-06-18:
On 17 June 2015 at 23:01, Courtney Powers <notifications@github.com> wrote:

> Ok I'll give it a shot unfortunately I just ordered the tssop package from
> digikey. I'll see if I can change the order to the new one package. Do you
> have an estimate for the new rev boards arrival?
>

Uh, sorry about that. I hoped to be a bit more useful for the general
population.
You can as well just edit the kicad board to use a tssop I guess (though
might need to fiddle with the routing). Maybe I should add one with double
footprints.

OshPark usually takes 2-ish weeks to fab.

squeegiefreak, 2015-06-18:
I think the bigger package will work better for most people in the long run. It might be good to have the double foot print option for people upgrading to the new library and pinout. Although the chips aren't that expensive.

hzeller, 2015-07-01:
Going to close this now. Did you receive your new boards already ?
Please re-open if you see any issues.

closed 2015-07-01 by hzeller