hzeller/rpi-rgb-led-matrix #35: led-image-viewer does nothing

felixmc, 2015-06-22:
I have a raspberry pi version 1 running latest Raspbian and a 32x32 LED matrix. All the "led-matrix" demos work as expected. I have the "libgraphicsmagick++1-dev" installed and "led-image-viewer" builds without any issues.

However, whenever I try to run "led-image-viewer", nothing happens on the LED matrix (stays completely off the whole time). I tried running it with a variety of images with different formats and dimensions. This is the output of running the command (which seems normal):  
[image]

Any ideas on why it might not work or what steps I can take to further debug this?

hzeller, 2015-06-22:
I don't see the output, looks like the image does not exist. Can you just copy/paste the terminal output ?

felixmc, 2015-06-22:
Sorry, not sure what happened to the image, here's a direct Dropbox link to it: https://www.dropbox.com/s/rqs3capnglcwabu/Screenshot%202015-06-21%2022.06.33.png?dl=0

hzeller, 2015-07-01:
Do you have a fresh checkout of the repository ? There was a brief phase when the image viewer code didn't work. Could you try with a fresh pull ?

felixmc, 2015-07-11:
pulled down again and rebuilt led-image-viewer and everything worked on the first try :)

closed 2015-07-11 by felixmc