hzeller/rpi-rgb-led-matrix #37: New and old library - connections

currentlysober, 2015-06-30:
Dear Henner,
I downloaded your source code and I tried to run it on 16x32 and 32x32 panels. However, I have some questions.
1. I downloaded the current source code. And I used Adafruit tutorial to connect matrix to RPi 2 ("connecting-a-16x32-rgb-led-matrix-panel-to-a-raspberry-pi"). I run text-example code and it works correctly, which is confusing because in your tutorial OE and CLK have changed. I keep connections as in Revision 1 (OE Pin3, CLK  Pin 5) - but in this case I will have a problem to connect 3rd display in parallel.
2. I am wondering whether there is some solution to make mirror display. Let's assume I have 1 32x32 display. I would like to show the same information on my 2nd 32x32 display. I do not want to use parallel/chain solution. Splitting cables would be the solution? From each output of Rpi2 I would lead out 2 cables and connect to each panel..? Maybe you tried it?
3. I already read solutions of running led matrix on Rpi and Arduino. I wonder whether there is some smaller/cheaper solution to run only one 16x32 panel? Maybe some Arduino clones?
Thank you in advance.
Best regards,
Daniel

hzeller, 2015-07-01:
> I downloaded your source code and I tried to run it on 16x32 and 32x32
> panels. However, I have some questions.
> 1. I downloaded the current source code. And I used Adafruit tutorial to
> connect matrix [...]
As long as you only do one panel, the pin-out is actually backward
compatible so that old wirings still work (I output output_enable on
multiple pins for instance).
As soon as you define  SUPPORT_MULTI_PARALLEL, this won't work anymore.

> 2. I am wondering whether there is some solution to make mirror display.
> Let's assume I have 1 32x32 display. I would like to show the same
> information on my 2nd 32x32 display. I do not want to use [...]
This should work. There might be trouble with too much load on the cables,
so maybe you need to define -DRGB_SLOWDOWN_GPIO.

> 3. I already read solutions of running led matrix on Rpi and Arduino. I
> wonder whether there is some smaller/cheaper solution to run only one 16x32
> panel? Maybe some Arduino clones?
>
Depends on what you want to do. The Arduino is very slow and you can barely
keep the refresh running; so if you want to change the output, then it
might be slow.
But maybe some STM32 could help - it is cheap, but powerful enough to do
all kinds of other things as well.

-h

currentlysober, 2015-07-01:
Thank you for your answers!

hzeller, 2015-07-25:
I guess I can close this now ?

closed 2015-07-25 by hzeller