hzeller/rpi-rgb-led-matrix #40: Porting to Cubieboard

aignaciors, 2015-07-03:
Hi hzeller,

Can you tell me if its possible to port your code for others platformos such as Cubieboard and BBB?

hzeller, 2015-07-03:
The gpio control is pretty specific to Raspberry Pi at  this point, but the actual framebuffer stuff is fairly independent.
It should be possible to port the functionality of gpio.{h,cc} to other platforms, but I haven't spent much time in abstracting enough away that that task is obvious.
The gpio.cc is about to get more complicated though, I am working on making it work with DMA, at which point the platform independence might be an interesting exercise; so if you want to get started on that, wait a week or so.

For beagleboard, there is a different project that uses the PRU http://beagleboard.org/project/octoscroller/

aignaciors, 2015-07-03:
Thanks for answer

Saij, 2015-07-03:
When I get my BBB maybe I'll try to port it or to extend the gpio.{cc,h} so that someone can define which system he want's to use.

hzeller, 2015-07-03:
Sounds good. Let me know when you start, because I am messing and simplifying the GPIO interface quite a bit currently.
BBB might be interesting, but it is not really a challenge because it has the PRU that allows for realtime stuff to run in parallel (though to be programmed in assembler. I am using it for some other project - a motion controller)

Saij, 2015-07-03:
Might take a while cause first I want to try to use LogiPI FGPA adapter board for the matrix.
(Goal is to make a very big display ^^)

closed 2015-10-19 by hzeller