hzeller/rpi-rgb-led-matrix #41: Overclocking and Flickering

Saij, 2015-07-03:
Hi,

played a little bit with overclocking and determined that overclocking result in flickering.

Here is my config.txt (only overclocking part)

    arm_freq=1000
    sdram_freq=500
    core_freq=500
    over_voltage=2
    force_turbo=1
    disable_splash=1
    boot_delay=0
    gpu_mem=16

RPi is stable and no problems with cyclictest. But as soon as I run led-matrix it will flicker. Even with -p 4

A test shows that after removing the overclock settings matrix just works fine again.
Just as information

Greetings
Saij

hzeller, 2015-07-03:
Interesting. In the master or the pwm branch ?

Saij, 2015-07-03:
Used the PWM branch

Saij, 2015-07-03:
Used also 3.18.9 RT5 kernel

hzeller, 2015-07-03:
Interesting. Maybe the transition between the two clock-domains is affected (the ARM clock domain and the PWM unit are separated by a queue).

hzeller, 2015-07-03:
Speaking of the RT kernel: did you see any difference ?

Saij, 2015-07-03:
Not really 
But it seems to Run smoother.
Without it you could sometimes see some flicker.
With it no flicker at all

beikimajid, 2018-03-08:
HI
I test raspbian with last RT_Kernel but see no diffrence and  I run led-matrix it will flicker.(p5 smd & 3 Rows & 6 Cols With passive PCB)
Are you sure that RT_Kernel is answer?

combs, 2023-04-23:
I have a feeling this may have changed in the intervening years?

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828