hzeller/rpi-rgb-led-matrix #47: New Boards

squeegiefreak, 2015-07-08:
I am having difficulties with the new board. I built two of them, and I can't get any of the examples to run, have you have any issues with yours? 

I have the new experimental pin out running, as well as parallel and I just get artifacting (while moving the cable.)

I built the second one with out the buffer caps and it updates, but only one 2 lines on the first display chain. I have a feeling something might be wrong with my raspberry pi. I am going to re-wire it to the new pin out on a breadboard to see.

Here are the parts I bought:

HC245
http://www.digikey.com/product-detail/en/0/568-3973-5-ND

100nf caps
http://www.digikey.com/product-detail/en/0/399-1096-1-ND

[image]

hzeller, 2015-07-09:
Looks good!

The new boards will only work with the experimental-pwm-bit branch currently. So when in the rpi-rgb-led-matrix directory, you need to type:
```
git checkout experimental-pwm-bit
```

(there is also a small chance that I messed up the board, as I haven't tested it yet...)

hzeller, 2015-07-09:
Ah, I see, you already use the new branch. I'll have to double-check the board tomorrow.

hzeller, 2015-07-10:
I got my boards, but have to wait for the HCT245 to arrive, but will have a more thorough look at the schematic to see if there is something suspicious.

Silly question: did you double-check that the cables are connected to the right side of the panel ? There is an input and output connector on the panel; and for the panels I have at home this means that the cable actually needs to 'turn around' when plugging in upside to the adapter (as you can see in this picture https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/img/three-parallel-panels.jpg ). There are often arrows printed on the board showing the 'direction of data flow'.

The fact that you get artifacts while moving things sounds like an unconnected input, which is why I am asking.

hzeller, 2015-07-10:
I tested the passive version of these boards and they work. My 245 arrive on Monday to test.

From the symptoms and the way the cable looks in your picture I suspect though connection on the wrong side.

squeegiefreak, 2015-07-13:
Some Clarification:

I have 8 of the same displays, all have the same results, I am only running one at a time to try and narrow down the issue (also using all three ports on the board). When I don't have a program running, then it only has random colors on two lines, and its not shifting around. 

if I run a program, I have multiple lines running at the same time, if I move the cable at all, I get crazy artifacting, if my hand gets anywhere near the back of the display board really. 

I have both the master and the experimental-pwm-bit branches, for the board, I'm running with experimental pin-out and parallel

at first I thought there might be something wrong with my Raspberry, cable, or display. But as soon as I plug into my breadboard with the old pin-out, led-matrix runs perfectly.

I also sanded off the nub on one of my cables, making it easily reversible, and if the cable is reversed, it acts the same as when the programs not running. I can make my cable look like your picture by swapping ends, as long as the red strip and the nub is on the right side it should be correct. 

Do you have the new pin-out somewhere so I can try it on my breadboard? I noticed the documentation still have the same pin-out.

Let me know if you have success with the board, and I will try and narrow it down, its possible I have noise or something being amplified by the 245's or something.

hzeller, 2015-07-14:
Soldered together a board, works:
[image]

Did you read my earlier response regarding my assumption ? Your last mail essentially confirmed that again.

I think you have connected the board on the wrong side of the LED panel (the LED panel has typically two connectors. One is input, one is output). I think you connected it to the output. There are two reasons why I think this
   - If the panels are lying upwards, the cables plugged into the adapter would need to be in a 'loop', pointing towards the chips when plugged in (see picture above). In your case they are plugged in so that they are straight out. That only happens if they are connected to the mirrored output side of the connector of the panel.
   - You get artifacts when you get close to the board (and in particular the 'back' of the board). This is as well a strong indicator: that is the side where the unconnected input is, which of course picks up all kinds of noise.

hzeller, 2015-07-14:
If it is wired up correctly, then double-checking the soldering joins (and that they don't gap) is of course the next. In your picture the direction of the chips look fine, but I can't see the soldering well; maybe there are some legs not entirely connected ?

squeegiefreak, 2015-07-14:
Thanks for the work on your end  to verify your design worked, I was able to reflow the board and got it working. I had the correct input side, and as long as the red side of the cable is opposite the USB, it should have the correct orientation. I attached a picture as an example. I appreciate you helping me get this working. I'll close this issue. Keep up the good work.

[image]

closed 2015-07-14 by squeegiefreak