hzeller/rpi-rgb-led-matrix #49: Different scan rates

mahatmajman, 2015-07-23:
Awesome library, really looking forward to getting stuck into it! One question I had though, is it possible to change the scan rate to use different modules? I noticed all of the outdoor LED matrix modules out of China these days seem to be going to a 1:4 scan rate instead of 1:8. The explanation I have had from two Chinese suppliers now is that 1:8 is the old standard and apparently 1:4 is better. Any idea how difficult it would be to modify the library to accept 1:4 scan rate modules? Thanks in advance,

hzeller, 2015-07-25:
Interesting, I only have seen 1:16 and 1:8 scan-rate so far. The 1:16 are the 32 row displays (as there are two banks with 16 rows each), the 1:8 are the 16 row displays. So a 1:4 would only have 8 pixel lines ? Or is it folded around in some way ?

1:4 is very simple to do, essentially the rows option needs to accept as well the value 8 (as this would be 2 times 4 rows).

1:4 scan will allow faster refresh rates for the same chain length, but as soon as you increase the chain length so that you have the same number of pixels per connector you probably end up having roughly the same speed.

Do you have a link to such a panel ? I'd like to see how they do the layout of the LEDs.

mahatmajman, 2015-07-25:
Here is one I found http://www.aliexpress.com/item/Very-Good-Leeman-P6-P10-Indoor-Outdoor-smd-digital-led-display-indoor-smd-led-display-dot/1931861877.html - it is 16 rows high so you have 4 rows lit at one time (twice brightness for outdoor displays), though it is the same 16 pin interface so I assume you latch in rows 1 and 5 on R1G1B1 and 9 and 13 on R2G2B2, so you would shift out twice the bits (192 x 2) per channel and effectively the C pin becomes redundant for a 16 row module. I have asked a supplier who has been particularly helpful to send through a timing diagram and if I can make sense of that I might get a sample in to try out. It has litterally been impossible to find a 1:8 P10 OUTDOOR RGB display though one supplier did offer to custom make them.

hzeller, 2015-07-25:
Yeah, don't worry, we'll get it to work :)

mahatmajman, 2015-07-27:
Cool! Here is a basic diagram from one I have been looking at - not that clear but shows at least how the data is shifted through on one of these modules. I'm going to try and get some more information on a couple of different modules for caomparison (this one was advertised as HUB75 but is actually HUB08!).
[image]

hzeller, 2015-07-28:
Interesting - this layout is very different from others, so you probably have to do your own pixel-mapping, so that pixel (x,y) actually shows up at the position you specify. I'll Reopen this for now to keep track of this once you actually got a physical sample board and we got it working.

mahatmajman, 2015-07-28:
The challenge is trying to find a standard, this is the only supplier who has been willing to share any shifting information so it's very hard to know if it is just their particular way of doing it or if it is becoming an industry standard. Have got one on the way so we'll see what happens when it turns up!

mahatmajman, 2015-08-24:
I finally got a couple of these in the mail and was really disappointed, the two modules from the same manufacturer each have different methods of addressing the LEDs (one is 16 x 32 and the other is 16 x 16 but the method for addressing is different between the two...) and there seems to be some sort of on-board PWM control. I couldn't get the details from the supplier and can't easily reverse engineer the modules (most of the electronics are potted in resin). I created a basic program which address the LEDs one at a time so I could hopefully determine the order but ended up turning on all LEDs in one column for the first 8 rows... Anyway based on the fact there does not seem to be a unified standard for the outdoor modules I've asked instead to have some custom made to the internal module circuit diagram.

closed 2015-08-24 by mahatmajman

laundryjuara2016, 2017-02-27:
Hallo,
Can you tell me , how to wire p10 rgb scan 1:4 with arduino? 

Thank u

Ellhassan213, 2018-03-07:
Hi,
Is it possible with this library to drive the Matrix statically instead of using the default scan mode?
I cannot achieve the refresh rate as high as i want, so i would like to get rid of that problem.

I am observing the LED lights through hammatsu orca scientific camera, unfortunately the pictures i am getting captures the frequency at which the LEDs are refreshed.

So is there a way to avoid refreshing the matrix?

Thanks very much

hzeller, 2018-03-07:
The panels need update all he time as they don't have any framebuffer; the Pi has to even do the PWM for each pixel @Ellhassan213 . Even if you only do one color mode without PWM, the panels need to be scanned through all available lines one-by-one, as only one line can be lit at a time. So no, these kind of panels do _not_ have a way to just statically display a picture.

Your best chance at a high update rate is to get some outdoor panel with 1:8 scanrate (32x32) or even better 1:4 scanrate (32x16 panels).Don't use 64x64 panels, they are slow.
Also, if you don't need too much color resolution, lowering the `--led-pwm-bits` can slightly increase the refresh rate.

As usual with watching LED screens with a camera, you need to adjust your exposure time to be longer than the refresh rate (you can see the refresh rate with `--led-show-refresh`). Typically, I'd make it double or triple; so if you have a refresh rate of 900Hz (typical with one of these 1:4 panels), then make the exposure not shorter than 1/300s.
Often, cameras are overwhelmed with the brightness of the LEDs and react with shortening the exposure time; you can prevent that with stopping down the aperture or with photographic gray filters.

b4rtek, 2022-11-04:
Hi, i have p4 panel with scan rate 1:10 p4 80x40. Best i can do i with parametrers ./demo -D 1 runtext.ppm --led-brightness=10 --led-rows=32 --led-chain=1 --led-cols=80 is duplicate by half screen and offset  10 pixels above and below ... a need help ..
I need to modify code to and scan rate 1:10?

davemaster, 2022-11-04:
> Hi, i have p4 panel with scan rate 1:10 p4 80x40. Best i can do i with parametrers ./demo -D 1 runtext.ppm --led-brightness=10 --led-rows=32 --led-chain=1 --led-cols=80 is duplicate by half screen and offset 10 pixels above and below ... a need help .. I need to modify code to and scan rate 1:10?
> 
> Ps. When i set --led-cols=160 then width match to pixels 1:1

What about a picture of your panel(s), from both sides

b4rtek, 2022-11-04:
this is how it looks like 
[image]
thanks for any advice :)

davemaster, 2022-11-04:
Good, now a picture from the back of your panel, more than one will be better, to see all the features of your panel in its ICs

hzeller, 2022-11-04:
You have a panel with more than 32 pixel rows, yet you give `--led-rows=32`. So if you have 40, you need to pass `--led-rows=40`
I'd start with that.

b4rtek, 2022-11-05:
@davemaster here is back of panel 
[image]
@hzeller that was first settings i try. Result: 
[image]
so i try all settings available in demo but unsuccess

davemaster, 2022-11-05:
Greetings,

try link below, in this, 80 = 16 x 5 chains, etc.. try all posibilities before a new post pls.

https://github.com/hzeller/rpi-rgb-led-matrix/issues/986#issuecomment-1000115417

b4rtek, 2022-11-05:
I was trying every combination. Also --led-chains=5 --led-cols=16 --led-rows=40 Its behave exactly same as --led-cols=80 --led-rows=40.

hzeller, 2022-11-06:
Did you wire up the `E` line @b4rtek ?

b4rtek, 2022-11-07:
@hzeller yes this variant i also test. 
Reference to this 
[image]
and 
[image]
But no results, no difference.
I notice pattern 
[image]
(parts are 10x20 pixels)
and start experimenting with lib/multiplex-mappers.cc modification adding new multiplexer.
It is possible to use 1 2 parts and extend to whole panel?

anandrajgupta, 2022-11-08:
i would suggest
`sudo ./demo --help`
and trying all possible combinations very carefully

b4rtek, 2022-11-08:
@anandrajgupta I've been doing this for a few days :)

b4rtek, 2022-11-08:
I wrote my multiplexer and this is results
[image]
[image]

Settings are no match to panel but its works
examples-api-use/demo -D 1 ./examples-api-use/runtext.ppm --led-brightness=10 --led-cols=**160** --led-rows=40 --led-pwm-lsb-nanoseconds=400 --led-multiplexing=19

and this is code

[image]

if someone has any advice how make it better i will be grateful
cheers.

ledvinap, 2022-11-08:
Can you post code as text, please?

You probably want stretch_factor=2 in base constructor.

b4rtek, 2022-11-09:
@ledvinap sure :) 
First thing... i try stretch_factor `P4Outdoor80x40() : MultiplexMapperBase("P4Outdoor80x40", 2) {}` but default behavior is to make from params --led-cols=80 --led-rows=40 to 160x20. So then i left it as is

Today i made two modifications in code
[image]
and now i can use --led-cols=80 --led-rows=40 and seems to work good

this is whole multiplexer
```
class P4Outdoor80x40 : public MultiplexMapperBase {
public:
  P4Outdoor80x40() : MultiplexMapperBase("P4Outdoor80x40", 2) {}

  void MapSinglePanel(int x, int y, int *matrix_x, int *matrix_y) const {
    int shift = ((y / 10) % 2);
    *matrix_y = y - shift * 10;
    *matrix_x = ((x * 2) + (shift > 0 ? 0 : 1));
  }
};
```
my only concern is low refresh ... ~35Hz

ledvinap, 2022-11-10:
> from params --led-cols=80 --led-rows=40 to 160x20. 

That is intentional - outdoor 80x40 is actually wired as 160x20 display (160bits in shift register / 20 rows (10 double rows)).
MultiplexMapper with stretch factor updates internal framebuffer representation, so that 160x20 is sent to display, but 80x40 is visible to user (canvas).

Your code is almost correct, but you must map second half of display y (x,20 -> x,10)
```
class P4Outdoor80x40 : public MultiplexMapperBase {
public:
  P4Outdoor80x40() : MultiplexMapperBase("P4Outdoor80x40", 2) {}

  void MapSinglePanel(int x, int y, int *matrix_x, int *matrix_y) const {
    int half =  y / (panel_rows_ / 2);  
    int shift = (y  / (panel_rows_ / 4)) % 2;
    *matrix_y = y % (panel_rows_ / 4) + half * (panel_rows_ / 2);
    *matrix_x = ((x * 2) + (shift ? 0 : 1));
  }
};
```

ledvinap, 2022-11-10:
Bottom half of display may be shifted, post image in that case

b4rtek, 2022-11-10:
@ledvinap your version of code works well but changes with stretch factor must stay

```
  virtual void EditColsRows(int *cols, int *rows) const {
    panel_rows_ = *rows;
    panel_cols_ = *cols;

    // *rows /= panel_stretch_factor_;  // <---- commented
    *cols *= panel_stretch_factor_;
  }

  virtual bool GetSizeMapping(int matrix_width, int matrix_height,
                              int *visible_width, int *visible_height) const {
    // Matrix width has been altered. Alter it back.
    *visible_width = matrix_width / panel_stretch_factor_;
    *visible_height = matrix_height; // * panel_stretch_factor_;  // stretch y <----- no multiplication
    return true;
  }
```

I was hope  your code will allow to back these changes and meaby makes less flickering

b4rtek, 2022-11-10:
With `  *rows /= panel_stretch_factor_;  ` uncommented and run with ` ./examples-api-use/demo -D 1 ./examples-api-use/runtext.ppm --led-brightness=50 --led-cols=80 --led-rows=40 --led-chain=1 --led-pwm-lsb-nanoseconds=300 --led-multiplexing=19`
[image]
and when i uncomment multiplication for `visible_height` the screen looks same but got errors on run 
[image]

ledvinap, 2022-11-10:
Strange .. when I run with P4Outdoor80x40  applied and your params, I get no PixelMapper errors
Can you try `git diff`?
```
class P4Outdoor80x40 : public MultiplexMapperBase {
public:
  P4Outdoor80x40() : MultiplexMapperBase("P4Outdoor80x40", 2) {}

  void MapSinglePanel(int x, int y, int *matrix_x, int *matrix_y) const {
    int half =  y / (panel_rows_ / 2);  
    int shift = (y  / (panel_rows_ / 4)) % 2;
    *matrix_y = y % (panel_rows_ / 4) + half * (panel_rows_ / 4);
    *matrix_x = ((x * 2) + (shift ? 0 : 1));
  }
};
```

ledvinap, 2022-11-10:
```
*matrix_y = y % (panel_rows_ / 4) + half * (panel_rows_ / 4);
```

I'm very sorry, I edited may post after posting it.

b4rtek, 2022-11-10:
No problem :) Thanks you. Now its works as expected.  `stretch_factor` modification are now not needed

ledvinap, 2022-11-10:
(you can skip `half` and use code from other mappers:)
```
 *matrix_y = ((y / (panel_rows_/2)) * (panel_rows_/4)
                 + y % (panel_rows_/4));
```

priya2212, 2023-01-17:
> I wrote my multiplexer and this is results [image] [image]
>
> Settings are no match to panel but its works examples-api-use/demo -D 1 ./examples-api-use/runtext.ppm --led-brightness=10 [...]

@b4rtek Can you please help me with the integration. My display doesn't work with the same multiplexer I have written. The display has 1:10 scan rate. I have raised issue here as well - https://github.com/hzeller/rpi-rgb-led-matrix/issues/1499

My whole development is on hold and you or anyone in this thread may guide me right direction. Please help. :(

commit c6fa34e5 (2015-07-25): o Make it possible to select 1:4 multiplexing. Fixes #49