hzeller/rpi-rgb-led-matrix #50: Running two instances don't work

Saij, 2015-07-23:
Hi,

just startet led-image-viewer with gif as daemon and then started the led-matrix demos.
But that isn't possible. The two processes are hanging with 90% CPU usage.

Maybe it's a good idea to implement a semaphore to disable the start of a second led matrix update thread.

Greetings
Saij

hzeller, 2015-07-23:
mmh, but wouldn't it be expected that things don't work if you start two instances that use the same hardware ? Not sure if it is too worthwhile protecting against this particular self-foot-shooting ...

Saij, 2015-07-23:
Yeah I expected that those two instances won't work together. But I could kill the daemon only with -9
And I think that maybe it would be good to stop people testing it out ^^

hzeller, 2015-07-25:
I think it is ok to leave people a little rope hidden so if they really want, they can hang themselve (or at least: hang the Pi).

closed 2015-07-25 by hzeller