hzeller/rpi-rgb-led-matrix #51: Adafruit HUB 75 Hat, -DRGB_CLASSIC_PINOUT option

magdesign, 2015-07-25:
Hi

I'm using an old Adafruit HUB75 Hat to drive the LED matrix.
So I must compile your code using the classic wiring option ( -DRGB_CLASSIC_PINOUT)

But I could not figure out where I have to write these -DRGB_CLASSIC_PINOUT arguments.
I guess somewhere in the Makefile.....

Thanks for your help!

hzeller, 2015-07-25:
If you are using the Adafruit HAT, then you actually need to give -DADAFRUIT_RGBMATRIX_HAT

Have a look in lib/Makefile, there are the options to switch on/off.

hzeller, 2015-07-25:
(and let me know if it works, because I don't have a Adafruit HAT to test, so I just implemented that pin-mapping looking at their description)

magdesign, 2015-07-25:
Thanks for your help.
I uncommented the option and tried to compile, but there is an error, here is what I get:

g++ -Iinclude -Wall -O3 -g -c -o demo-main.o demo-main.cc
make -C lib
make[1]: Entering directory '/home/pi/rpi-rgb-led-matrix/lib'
g++ -I../include -Wall -O3 -g -fPIC -DADAFRUIT_RGBMATRIX_HAT -c -o gpio.o gpio.cc
g++ -I../include -Wall -O3 -g -fPIC -DADAFRUIT_RGBMATRIX_HAT -c -o led-matrix.o led-matrix.cc
g++ -I../include -Wall -O3 -g -fPIC -DADAFRUIT_RGBMATRIX_HAT -c -o framebuffer.o framebuffer.cc
framebuffer.cc: In constructor ‘rgb_matrix::internal::Framebuffer::Framebuffer(int, int, int)’:
framebuffer.cc:49:1: error: uninitialized member ‘rgb_matrix::internal::Framebuffer::parallel_’ with ‘const’ type ‘const int’ [-fpermissive]
framebuffer.cc: In static member function ‘static void rgb_matrix::internal::Framebuffer::InitGPIO(rgb_matrix::GPIO*, int)’:
framebuffer.cc:83:10: error: ‘struct rgb_matrix::internal::Framebuffer::IoBits::<anonymous>’ has no member named ‘output_enable_rev1’
framebuffer.cc:83:38: error: ‘struct rgb_matrix::internal::Framebuffer::IoBits::<anonymous>’ has no member named ‘output_enable_rev2’
framebuffer.cc:84:10: error: ‘struct rgb_matrix::internal::Framebuffer::IoBits::<anonymous>’ has no member named ‘clock_rev1’
framebuffer.cc:84:30: error: ‘struct rgb_matrix::internal::Framebuffer::IoBits::<anonymous>’ has no member named ‘clock_rev2’
framebuffer.cc:115:27: error: ‘struct rgb_matrix::internal::Framebuffer::IoBits::<anonymous>’ has no member named ‘output_enable_rev1’
framebuffer.cc:116:31: error: ‘struct rgb_matrix::internal::Framebuffer::IoBits::<anonymous>’ has no member named ‘output_enable_rev2’
framebuffer.cc: In member function ‘void rgb_matrix::internal::Framebuffer::DumpToMatrix(rgb_matrix::GPIO*)’:
framebuffer.cc:324:23: error: ‘struct rgb_matrix::internal::Framebuffer::IoBits::<anonymous>’ has no member named ‘clock_rev1’
framebuffer.cc:324:56: error: ‘struct rgb_matrix::internal::Framebuffer::IoBits::<anonymous>’ has no member named ‘clock_rev2’
framebuffer.cc:333:14: error: ‘struct rgb_matrix::internal::Framebuffer::IoBits::<anonymous>’ has no member named ‘clock_rev1’
framebuffer.cc:333:38: error: ‘struct rgb_matrix::internal::Framebuffer::IoBits::<anonymous>’ has no member named ‘clock_rev2’
Makefile:61: recipe for target 'framebuffer.o' failed
make[1]: *** [framebuffer.o] Error 1
make[1]: Leaving directory '/home/pi/rpi-rgb-led-matrix/lib'
Makefile:23: recipe for target 'lib/librgbmatrix.a' failed
make: *** [lib/librgbmatrix.a] Error 2

hzeller, 2015-07-25:
Uh, give me a minute to fix that ...

hzeller, 2015-07-25:
Ok, do a pull and try again ...

magdesign, 2015-07-25:
Thanks for your hi-speed support!

Compilation works fine and also the ./led-matrix runs smoothly with 4 LED matrix on a HUB75 Adafruit Hat.

I think you can close the issue.

hzeller, 2015-07-25:
Cool.

closed 2015-07-25 by hzeller