hzeller/rpi-rgb-led-matrix #52: Building Python Library

ThomTrab, 2015-07-27:
Hello,
Thank you very much for all your work, I can not do it but I'm happy to use the python library. But I'm not able any more to build the python library based on the lastest version of the repo on a Raspberry Pi.
Here under the error message:
```
make -C python build
make[1]: Entering directory '/home/pi/tweety_pi/matrix/python'
cython --cplus -o rgbmatrix/core.cpp rgbmatrix/core.pyx

Error compiling Cython file:
------------------------------------------------------------
...
        uint8_t g
        uint8_t b

    cdef cppclass Font:
        Font() except +
        bool LoadFont(const char*)
                               ^
------------------------------------------------------------

rgbmatrix/cppinc.pxd:52:32: Expected ')', found '*'

Error compiling Cython file:
------------------------------------------------------------
...
cimport cppinc

cdef class Canvas:
    cdef cppinc.Canvas *__getCanvas(self) except +
        ^
------------------------------------------------------------

rgbmatrix/core.pxd:4:9: 'Canvas' is not a type identifier

Error compiling Cython file:
------------------------------------------------------------
...

cdef class Canvas:
    cdef cppinc.Canvas *__getCanvas(self) except +

cdef class FrameCanvas(Canvas):
    cdef cppinc.FrameCanvas *__canvas
        ^
------------------------------------------------------------

rgbmatrix/core.pxd:7:9: 'FrameCanvas' is not a type identifier

Error compiling Cython file:
------------------------------------------------------------
...

cdef class FrameCanvas(Canvas):
    cdef cppinc.FrameCanvas *__canvas

cdef class RGBMatrix(Canvas):
    cdef cppinc.RGBMatrix *__matrix
        ^
------------------------------------------------------------

rgbmatrix/core.pxd:10:9: 'RGBMatrix' is not a type identifier

Error compiling Cython file:
------------------------------------------------------------
...
cdef class FrameCanvas(Canvas):
    cdef cppinc.FrameCanvas *__canvas

cdef class RGBMatrix(Canvas):
    cdef cppinc.RGBMatrix *__matrix
    cdef cppinc.GPIO *__gpio        ^
------------------------------------------------------------

rgbmatrix/core.pxd:11:9: 'GPIO' is not a type identifier

Makefile:43: recipe for target 'rgbmatrix/core.cpp' failed
make[1]: *** [rgbmatrix/core.cpp] Error 1
make[1]: Leaving directory '/home/pi/tweety_pi/matrix/python'
Makefile:49: recipe for target 'build-python' failed
make: *** [build-python] Error 2
```
I try on my laptop and it's OK, even if doesn't seem to use cython in the process
Thank you,

Saij, 2015-07-27:
Which version of cython are you using currently?
Will take a look why the Makefile will regenerate the cpp files instead using the ones provided.

ThomTrab, 2015-07-27:
I'm using Cython version 0.15.1

hzeller, 2015-07-27:
The Makefile looks for timestamps - when the source is newer, it will generate the target. However, after checkout, the timestamps are probably pretty close together (and randomly so), so if there is a direct dependency cython-input -> *.cpp and then there is some other dependency that lists that *.cpp, then make fill follow the dependencies and possibly re-build the *.cpp.

Maybe the cython-build step should be less formulated as a dependency in that case but just as an explicit step
generate-code:
    # command line that makes it happen

Let me know if you need help with the makefile @Saij

Saij, 2015-07-27:
Ah I see.
Will change the makefile as you suggested.

Saij, 2015-07-27:
OK created a pull request and moved the generation of the code from cython to a separate command.
Only thing is, I couldn't test it 'cause my PI is currently not working.

hzeller, 2015-07-28:
This should be working now. @ThomTrab, can you do a fresh pull and try again ?

ThomTrab, 2015-07-28:
@hzeller I don't have the PI in hand. I will do this as soon as I have 
it. This should be beginning of next week. I keep you posted

Thank you very much both of you @hzeller and @saij for your very 
reactive answer and solution.

ThomTrab, 2015-08-03:
Work like a charm

closed 2015-08-03 by ThomTrab