hzeller/rpi-rgb-led-matrix #56: Running led-matrix as daemon

ts4pi, 2015-08-02:
Hi,

thanks to Your code I got my 32 x 256 Matrix running. Now I'd like to start led-matrix as a daemon but I've got no idea how send new images to the daemon. I tried to copy ppm-files to /dev/mem but this causes my pi to crash. 
Could you give me an example ?
Thanks in advance
Thomas

hzeller, 2015-08-03:
The demos don't do image changing.
But you can write your own program that does that - the demos are there to get you started, but you still have to write your own code if you have different uses; it comes with an easy-to-use API.

hzeller, 2015-09-13:
You could as well try to use the PixelPusher implementation, which provides a standard protocol to send images remotely to the panel. Or you could use it as an inspiration for your own experiments:

https://github.com/hzeller/rpi-matrix-pixelpusher

hzeller, 2015-09-13:
(closing for now)

closed 2015-09-13 by hzeller