hzeller/rpi-rgb-led-matrix #57: Different colors are displayed

k54821, 2015-08-19:
Hi.

We are testing the program connected with four 32x32 LED Display Panel X.

But the color looks differently in some parts of panal.

I think It's only appear from 32x32 Panel. What is the problem is it ?

gif url : http://www.cartoondollemporium.com/forum/pics/cdegraphics/12473213481194989982.gif
Test Video : https://www.youtube.com/watch?v=EYMj5Fbc2as&feature=youtu.be

Saij, 2015-08-19:
Looks like the gif is mirrored.

makifbuz, 2015-08-19:
Hi, some panels green and blue pins are reversed. I think yours too. You should change the gpio description between green and the blue pins or the hardware connection. For e.g. you should connect rpi g1 pinout to b1 and so on.

k54821, 2015-08-20:
makifbuz, Everything is great after we change G1, G2 and B1, B2 pins, I'm really appreciate it.

closed 2015-08-20 by k54821

makifbuz, 2015-08-20:
I am glad to hear that.