hzeller/rpi-rgb-led-matrix #58: Swapped blue and green

jblang, 2015-08-20:
I have a 32x32 RGB matrix purchased from Adafruit in July 2013. I believe it is a 4mm pitch; it was the only one being sold at the time at a price of $120.  The Blue and Green pins seem to be swapped from what you have documented and coded for.  When I wire the matrix as instructed and use the stock code, blue and green are swapped. I have verified this using the text demo by selecting 100% green or 100% blue text, and by looking at the colors in the scrolling runtext.ppm demo where the raspberry's leaves are pink instead of green and the smiley face is magenta instead of yellow.

Either swapping the G1 <-> B1 and G2 <-> B2 wires, or swapping the variable names in the IoBits structure in framebuffer-internal.h fixes the issue. I had initially swapped the blue and green elements in IoBits structure, but when I redid my wiring for the hardware PWM you recently added, I opted to swap the blue and green wires so I don't have to maintain a separate branch of your code anymore. Has anyone else reported this?  If it's rare, I am not sure it is worth an extra ifdef in the code but it may at least be worth a footnote in the documentation.

As a side note, I wrote up the wiring technique I used in my blog here: http://adventuresinelectronics.blogspot.com/2015/02/leds-part-deux.html.  If you want to link it from your documentation, it may be preferable to use my wiring method for people who don't know how to solder or have the necessary equipment.

k54821, 2015-08-20:
we change G1, Bl and G2, B2 pins of 32*32 Panel, Then the problem is resolved. But is it possible this problem deal with on the program option ?

hzeller, 2015-09-01:
Interesting, this is the first time I hear that blue and green is swapped - but of course there are all kinds of 'almost' compatible LED panels out there, so it is not entirely strange.

I'll think of a way to re-map these with a #define

hzeller, 2015-09-13:
Do you have a source of where you got the particular panel ? Maybe I should buy one to test compatibility.

jblang, 2015-11-03:
Adafruit in July 2013. I believe it is a 4mm pitch.

commit e655b735 (2015-12-02): o Provide a way to swap green and blue. Some panels out there are
  broken that way.
  Fixes #58
closed 2015-12-02 by hzeller