hzeller/rpi-rgb-led-matrix #59: Clock frequency

mahatmajman, 2015-08-24:
I have some Chinese modules that work relatively well (they are 16 x 32 modules), but with more than 4 in a chain I start to get lots of noise in the display  (looks like some of the pixels are ending up in the wrong place - but colours are staying consistant) regardless of PWM bits (seems to be the same whether 1 or 11 bits are set for PWM). The more LEDs I have switched on in the modules the worse the noise seems to get. I have spoken with the supplier and they have said the max clock frequency is 24MHz... Dumb question - what is the clock frequency set to with this library? The only references I could find in the library indicated 500MHz but that seems to be related to PWM. Is there a way that the clock frequency can be adjusted? I might see what I can see with the oscilloscope tonight...

mahatmajman, 2015-08-24:
Actually, just read your readme about cable sizes and voltage drop, increased the supply by 0.5V to test and problem solved. I didn't realise these were so sensitive to supply voltage level! Now to find some larger cable and capacitors...

mahatmajman, 2015-08-24:
I'm not using the makefiles as I am using this library for a custom application, just found that adding #define RGB_SLOWDOWN_GPIO to the gpio.h file (instead of modifying the lib makefile) also makes a big difference. I am able to drop the voltage as low as 3 volts with no pixel noise now. Seems to run very stable though I'm sure I have sacrificed frame rate for it!

closed 2015-08-24 by mahatmajman

hzeller, 2015-08-24:
Yes, RGB_SLOWDOWN_GPIO is the way to go in these cases. Framerate is a little lower, but now half. If you are interested, there is another DEFINE that shows framerate so you can experiment with both options.