hzeller/rpi-rgb-led-matrix #60: Trying to run ThreadedCanvasManipulator again

XPerianer, 2015-08-24:
I don't know if this is a bug or intended, but if you try to stop ad start a thread with ThreadedCanvasManipulator you run into problems. 
If you try: 
    demo->Start();

hzeller, 2015-09-01:
mmh, I think I never thought about re-starting a new thread; usually these things are meant to be started once and used to the end, thus this was not explicitly done. It could probably be implemented if there is such need.

But what do you intend to do ?

XPerianer, 2015-09-11:
Hello, thank you for the reply, 

my intention is to do a project where you can remotly control the matrix and it's output. 
For the program, this means it waits for input(key press on a remote device over a socket) and then displays a specific animation. Then it waits for the next key press and should now display a diffrent thing. I got it to work with the little "workaround" that you create a new instance of the class, but for different animations it would be better to be able to start and stop the thread as you whish to do.

Sincerly, 

Xperianer

hzeller, 2015-09-13:
The idea of the thread class is that it runs the Run() method. Once it exits, it is done. This is pretty much how all Thread class implementations work, they are rarely meant to be re-used. Of course, this could be implemented.
It sounds like you want to solve a different problem: you want to wait for something (etc) and then continue with the routine.

For that, I would suggest to use a Mutex and a condition variable to do this - it is a lot cleaner.
(of course, if you already have things running re-creating an object this is fine as well).

commit 6d2383c6 (2015-09-13): o Make ThreadedCanvasManipulator restartable (fixes #60)
closed 2015-09-13 by hzeller