hzeller/rpi-rgb-led-matrix #61: Blue and Red rows

roykolak, 2015-08-26:
Using a Pi B+ w/ the new wiring diagram to drive a 16x32 display. When I run a demo, any demo, I see a row of red in the middle and a row of blue at the top. Not sure why this is, seen this behavior before?

roykolak, 2015-08-27:
Ah! note: if you mess w/ the header files after you `make`, you need to `make clean` first.

closed 2015-08-27 by roykolak

hzeller, 2015-08-27:
Interesting, what did you have to change in the header file to make your display to work ? My initial thought was that a simple -r 16 would've been sufficient ?

roykolak, 2015-08-27:
I wired up my displays via the adafruit guide and updated the header file to support the older layout, but was having some issues. So I tried the newer wiring diagram and just did a `make` and was experiencing the blue and red line issue. The `make clean` then `make` fixed it.

cheragnb, 2016-04-14:
I have the exact same problem with the exact same output. Im using the wiring layout shown here on github . https://github.com/hzeller/rpi-rgb-led-matrix

I have no clue how to fix it , help please ?

hzeller, 2016-04-14:
Interesting, this must be a slightly different way your particular hardware works.

Did you try to change the GPIO slowdown ?

cheragnb, 2016-04-14:
Yea the GPIO slow down worked. But, now the screen works on the demos, but
not when I try to project my monitor on to it. Even the test shows just a
blank screen rather than the matrix indexes in white.

Thanks for replying so soon , your library and documentation has been a
great help.

-Cherag

roykolak, 2016-04-14:
guessing `make clean` didn't fix it for you?

cheragnb, 2016-04-14:
Nope, not for projecting the screen. For the demos yes.

hzeller, 2016-04-14:
How do you try to project your screen @cheragnb ? Did you write software for it ?

cheragnb, 2016-04-14:
No i followed the adafruit instructions. After setting up the matrix.cfg
matrices and setting panel rotations, the screen just turns blank.