hzeller/rpi-rgb-led-matrix #63: led matrix flicking.

k54821, 2015-09-01:
it is connected 2 lines which has 6ea of 32 x 16 led pannel. Total pixel is 192x32. When i check end of left side, there is flicking. even i tried to use rpi-rgb-led-matrix adapter, the result was totally same. please inform your solution.

hzeller, 2015-09-01:
Depending on the speed of the LED panels, it might help to define RGB_SLOWDOWN_GPIO (see lib/Makefile)

k54821, 2015-09-02:
We also tried to use RGB_SLOWDOWN_GPIO and Active-3 adapter.
But it's not work. Please check the attached video.
Let us know the another solution.

https://www.youtube.com/watch?v=g7Ul_Z-y7C8

hzeller, 2015-09-03:
On 1 September 2015 at 21:40, k54821 <notifications@github.com> wrote:

> We also tried to use RGB_SLOWDOWN_GPIO and Active-3 adapter.
> But it's not work. Please check the attached video.
> Let us know the another solution.
>
> https://www.youtube.com/watch?v=g7Ul_Z-y7C8
>

Ha, that literally looks like 'Heartblead'.

So looks like these panels still can't deal with the speed, and maybe you
need to add more 'slowdownness'
in line 47 and 56 of this file:

https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/gpio.h#L47

k54821, 2015-09-04:
First, thank you for your help,
But I can't understend what the 'slowdownness' mean is.
Could you give me an Example?

hzeller, 2015-09-05:
I mean to repeat another line, e.g.
#ifdef RGB_SLOWDOWN_GPIO
    *gpio_set_bits_ = value;
    *gpio_set_bits_ = value;  // one more.
#endif
Same for the clr bits.

k54821, 2015-09-10:
Thank you for your help,
It's doing nicely, perfact~! Thank you.

hzeller, 2015-09-13:
I have now, in commit 246becad2e3f4cff091b61f6795b78ac0e84cc5a, added a way to further slow down without manually modifying the code.
Instead of -DRGB_SLOWDOWN_GPIO, you can say -DRGB_SLOWDOWN_GPIO=2 now, and it will add the one additional slowdown.

closed 2015-09-13 by hzeller