hzeller/rpi-rgb-led-matrix #68: Use parallel LED-Ring and LED Matrix

heinydo, 2015-10-09:
Hi,

hzeller, 2015-10-09:
It might well be that the initialization between the different PWM channels (this library uses channel 0 with GPIO 18 on Pin 12) conflict.
Since I have not used the other channel in parallel, I never observed that. If you want, you can have a look at the BCM datasheet and the codes initializing the channel in both libraries and try to see what is happening.

What this library is doing, for instance, is to modify the base frequency of the internal timer to do get the right update rate; if this is a shared resource between the PWM channels, then this will mess up the other channel.

noethno, 2015-10-19:
Hi,
I am working together with heinydo on this project.
We still got the parallel problem with ring and matrix.
We think this could be a problem, mmap accessing /dev/mem because the 
ringlibary is accessing /dev/mem too.
Could the libaries block each other while accessing the same file with read/write attributes?
Or is it possible they write into the same memory space? 

Here the link to the related code lines in the matrix libary: https://github.com/hzeller/rpi-rgb-led-matrix/search?utf8=%E2%9C%93&q=%2Fdev%2Fmem

And here the related code lines in the ring libary: https://github.com/jgarff/rpi_ws281x/search?utf8=%E2%9C%93&q=mmap&type=Code

Best regards.

hzeller, 2015-10-19:
/dev/mem is probably not a problem, every process can open and it.  Writing to the GPIO pins typically does not interfere either, as these accesses are masked.

Did you investigate the assumption of interfering time-bases for the PWM channels ? The matrix needs to change the base frequency to 500Mhz (I think the default is 100Mhz) - so if the base frequency of the PWM subsystem controlls all channels, then your timing could be 5x too fast if these interfere; which you could fix by making the time constants 5x as much in the NeoPixel thing (the ws28xx are really sensitive to timing).

You give little information _what_ is not working. Are there any error messages ? Does the matrix stop when you start the NeoPixel ? Or only the NeoPixel when the Matrix starts ? Do they both stop ? What do you see with an oscilloscope on the pins ? Do you see change in Frequency  ?

You need to look at this systematically. And get the datasheet and read how these subsystems work. You can just refer to the BCM2835 as these things seem to be the same compared to the 2836 (Pi)
https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2835/BCM2835-ARM-Peripherals.pdf - in section 9.1 for instance it describes that both channels use the same clock.

So: systematically debug this, pull out an oscilloscope and look at the signals. I suspect you just need to increase the time the ws28xx library uses by 5x.

hzeller, 2015-10-29:
Did you figure it out, something you can give as advice for people that stumble into the same problem ?
Was it the time constant needing to be 5x ?

closed 2015-11-12 by hzeller