hzeller/rpi-rgb-led-matrix #69: text aliasing of led-matrix -D 1 or 2 on 32x32

yarikoptic, 2015-10-10:
Thanks for the project and detailed HOWTO -- getting 32x32 (6mm pitch) from adafruit  running (using old PC power supply for 5V) was quite easy (ultimately need to chain 4).  All demos seems to work nicely **but** the scrolling text -- my primary target.  What could lead to text aliasing such as shown at http://www.onerussian.com/tmp/LED-text-aliasing.jpg .  Usually it just aliases to the right (up to 3 LEDs) and if there is an extended number of light up LEDs seems sometimes happen to the left LEDs as well.  The effect of right aliasing is independent from direction of text scrolling (-D 1 vs 2) -- guaranteed aliasing to the right.  I have tried rebuilds with varying values for `DEFINES+=-DRGB_SLOWDOWN_GPIO=1` which seemed had no effect.  Static text (text-example) shows beautifully.
Any advice on where should I dig to manage to avoid having such aliasing effect?

yarikoptic, 2015-10-11:
just to cross-link: posted question also to https://forums.adafruit.com/viewtopic.php?f=47&t=81885  Cheers!

hzeller, 2015-10-11:
I have found that this is a hardware issue with some panels, which cannot be prevented in software as far as I know.
Sometimes it helps to slow down the GPIO writes (with -DRGB_SLOWDOWN_GPIO=1 (or even 2) - see lib/Makefile).
Maybe increasing or decreasing the voltage to the panel a little bit can help as well.

yarikoptic, 2015-10-11:
Thanks! That sucks though... By some panels you meant since models (thus I should expect the same in the other 3 I have) or particular instances?
I did try to show GPIO  down with that compiler flag without any effect... BTW, should slowing it down decrease the refresh rate? (I have enabled that reporting) in my case it also wasn't effected

hzeller, 2015-10-11:
Yeah, with 'some panels' I mean panels from a particular manufacturer. But it is hard to predict, often various manufacturers use the same type of regulators etc. I have not measured yet on the electrical level what is happening, but suspect that they have some cross-talk in their 1:8 or 1:16 multiplexer.

Slowdown will affect the refresh rate, but it only really has an impact with longer chains, so with a single panel, you might not see anything.

yarikoptic, 2015-10-11:
Gotcha... I wonder, if there was a flow up/down, would I observe such an effect on any direction? :-)

hzeller, 2015-10-11:
You should actually try this with static, simple geometric things, such as a single pixel or a few pixels to see which other pixels might be affected.

yarikoptic, 2015-10-11:
Well, Static text (text-example) shows beautifully.

hzeller, 2015-10-11:
Ah, then it is not the problem I am thinking of.
If you have moving artifacts, you should try to use SyncOnVSync() with double buffers.

yarikoptic, 2015-10-11:
I guess you have meant SwapOnVSync (I see no SyncOnVSync) but then not sure how to try since that one is already used in the ImageScroller::Run in preparation  the next screen

hzeller, 2015-10-11:
Ah, yes, that is what I meant - a typo.

The API documentation should be explaining what is happening. If you think of standard animation, this is essentially double-buffering.
You create an off-screen buffer, fill that, and then swap it with the other buffer. Then you can write on the new offscreen buffer (that SwapOnVSync() returned) and start over.

yarikoptic, 2015-10-11:
Rright -- but isn't that what is already happening in the ImageScroller::Run or you suspect that vsync comes when new buffer (offscreen_) gets plotted before loop completes?  not sure though how that would cause aliasing in both directions... or always to the right regardless of the direction of animation -- or am I still misunderstanding?

hzeller, 2015-10-11:
Well, if you already use the ImageScroller, then it sounds like it is just an optical artifact where your eyes just average adjacent images.

(I wouldn't use the ImageScroller when you just want to scroll text, it might be easier to implement that using the text features).

(also I presume, you already made sure that your input image is crisp without aliasing ? Do  you have the input image that you can attach to this bug ?)

yarikoptic, 2015-10-11:
neh -- wasn't just optical artifact ;-) even phone photo somewhat registered that.  Although I might indeed play a bit with scroll_ms and see what effect that one would have.
I left my contraption at work, but indeed I did check the images and they did look crisp.  Even though I am 99% sure I did (and just visually), I will check tomorrow again to get 100% sure  ;)
I thought to scroll images at few times (primarily text though): actually I might need customization anyways if it would work: to keep static left 32x32 matrix (I will chain them anyways -- can't mess with wiring more) with qrcode (if shows to be scannable with the phone) while scrolling text in the right 3 32x32 matrices.

Saij, 2015-10-12:
I also encountered this phenomen.
Maybe trying to blank out the active buffer before switching to the new one helps?

yarikoptic, 2015-10-12:
I am truly sorry for wasting your (and my) time.  It was indeed fonts antialiasing resulting in those slightly lit LEDs... problem solved!!! THANK YOU BUNCH for your patience.

closed 2015-10-12 by yarikoptic

hzeller, 2015-10-12:
Might be good if you updated the Adafruit thread as well, so that people don't unnecessarily have to scratch their heads there.

yarikoptic, 2015-10-12:
Thanks for the reminder!  did it now. Cheers!

yarikoptic, 2015-10-12:
ah -- now I recall why I was sure it wasn't just font rendering in the image... because I have tried provided runtext*.ppm and observed the same effect... but apparently font is aliased in them now that I checked ... that is why I was 99% sure ;)