hzeller/rpi-rgb-led-matrix #70: Use of led reproduce

SungJinChoi, 2015-10-12:
Hi,I would like to use to clone the display without chains.
But the image does not come out normally.

In the video demonstration
> ./led-matrix -D1 Runtext16.ppm -r16 -m 5000
Use the command output results.

I would like to thank the replication means that there is no solution. Thanks.

a demonstration: 
https://www.youtube.com/watch?v=MIZ-7PZfKMg

hzeller, 2015-10-13:
If you want to clone the image in a chain, you have to clock each row twice so that it makes it to the end of the second display.
You have to manually modify the code for that, in lib/framebuffer.cc

```
--- a/lib/framebuffer.cc
+++ b/lib/framebuffer.cc
@@ -344,16 +344,18 @@ void Framebuffer::DumpToMatrix(GPIO *io) {
 
     // Rows can't be switched very quickly without ghosting, so we do the
     // full PWM of one row before switching rows.
     for (int b = kBitPlanes - pwm_to_show; b < kBitPlanes; ++b) {
+      for (int k = 0; k < 2; ++k) {
         IoBits *row_data = ValueAt(d_row, 0, b);
         // While the output enable is still on, we can already clock in the next
         // data.
         for (int col = 0; col < columns_; ++col) {
           const IoBits &out = *row_data++;
           io->WriteMaskedBits(out.raw, color_clk_mask.raw);  // col + reset clock
           io->SetBits(clock.raw);               // Rising edge: clock color in.
         }
+      }
       io->ClearBits(color_clk_mask.raw);    // clock back to normal.
 
       // OE of the previous row-data must be finished before strobe.
       sOutputEnablePulser->WaitPulseFinished();

```

SungJinChoi, 2015-10-13:
The letters are flashing around a little bit, but that was replicated. Thank you.

hzeller, 2015-10-13:
Writing the same value multiple times is the simplest way to slow down the GPIO output.
(also, it looks like  you have an old version of this code)

SungJinChoi, 2015-10-15:
Oh, I see. Thank you.
I found two problems today confirmed that the first 16 pixels over to see you again replicated know when you output to a sparkling sensation severe that yirang second 32 pixels around the characters are pushed out to the right by one pixel I get from the source is there a way?

hzeller, 2015-10-15:
I don't really understand what you mean, can you reformulate ?

hzeller, 2015-10-29:
Is this done, or can you give a better description of what you need ?

closed 2015-11-06 by rami2820