hzeller/rpi-rgb-led-matrix #73: 74HC245 vs. 74HCT245

urshofer, 2015-10-20:
Hi -
I got some LED Boards from a previous installation left over and I'd like to use them with the RPi2. Checking the chips to make sure it runs with 3.3 Volt i saw the type: "ChipOne ICN74HC245TS". Does anybody know if this is a HC or HCT type chip? Found a supplier on aliexpress, but there ain't no datasheets around.
Thanks for any hints,
best
Urs

[image]

hzeller, 2015-10-20:
If it says HC245, it is a HC chip - the letters between the 74 and 245 are relevant here.

So it will not work with the 3.3V output of the RPi (or only very unreliably; you _could_ experiment in reducing the voltage of the panel to something like 4V, then it might properly read the digital signal - but the colors are off as the internal LED current limiters don't work anymore properly).

So you either have to rework the panel and replace the chip, or build an adpater (see adapter/ directory), with a HCT or AHCT chip (recommended).

urshofer, 2015-10-20:
Thanks! I'll go fot the adapter - and thanks for your excellent work!

closed 2015-10-20 by urshofer