hzeller/rpi-rgb-led-matrix #75: 16x32 panel, 2x4 rows don't light up

jmakacek, 2015-10-22:
Hi,

I'm just getting into Pi development and have run into an issue trying to display stuff on 16x32 led panel using hzeller/rpi-rgb-led-matrix library.  I apologize in advance if I don't have the right terminology for this, just getting into it.

The panel/code seems to work except the LED rows light up in a weird way.  Counting rows from the top of the display
4 rows light up and display properly
4 rows dark (all the time)
4 rows light up and display properly
4 rows dark (all the time)

Hardware:
- Raspberry Pi 2
- 74AHCT125 - Quad Level-Shifters to boost voltage from 3.3 to 5 on all wires except GND (3 chips, 12 wires total) - (https://www.adafruit.com/product/1787) - so, 12 wires are "shifted" to 5V, GND is not, and not using pin 22 (D) since it's only 16x32 board
- 16x32 led matrix panel (http://www.adafruit.com/products/420)
- Breadboard and a whole bunch of jumper wires

I am using the rpi-rgb-led-matrix library without modification (download and make), and have it wired the "new" way, the way described in the wiring diagram on this page: https://github.com/hzeller/rpi-rgb-led-matrix

It's worth mentioning that I had the whole panel lit up before with the "old" way of wiring it (had other issues) so I'm fairly certain it's not a faulty panel.  I also checked and re-checked the wiring and it looks good.  I'm pretty sure the level-shifters are good as I measure the input and output voltage on them before.

Anyway, any example/command I run, it is displayed the way I described above: rows 1-4 and 9-12 display properly while rows 5-8 and 13-16 do not.

I was wondering whether you have come across an issue like this before and/or have any suggestions.

Any help is greatly appreciated as I'm getting quite frustrated at this point.

Thanks in advance.

jmakacek, 2015-10-22:
Hi,

Here is a sample of what it looks like: https://drive.google.com/file/d/0B4vWHIIVWt9nTjE0LTZJMDVaU28/view?usp=sharing

Command: sudo ./led-matrix -D 4 -b 25 -r 16

Thanks.

hzeller, 2015-10-22:
Looks like your 'C' signal is not making it through - maybe double check the cabeling on these.

jmakacek, 2015-10-22:
Wow!  Quick response and spot on, thank you so much!

jmakacek, 2015-10-22:
Should have added, turned out wiring was correct (pin 18 -> C) but it was getting lost at some point.  Replaced one of the chips (level shifter) and replaced the wires, used different holes on the breadboard (just in case) and it started working.

Now to figure out how to deal with artifacts...  Thanks again!

closed 2015-10-22 by jmakacek

hzeller, 2015-10-22:
What artifacts are you getting ? Sometimes, signals are too fast with  the Raspberry Pi2, so the SLOWDOWN option (see lib/Makefile) might help.

(and for more panel fun with three chains in parallel and less breadboard adventures, you can build yourself a https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3 )

jmakacek, 2015-10-22:
When I run scrolling text: sudo ./led-matrix -t 10 -D 1 -r 16 runtext16.ppm -b 25
I get these "funky" LED "flashes" of different colors around the text, etc.  

Random LEDs lighting up (more towards the end of the video): https://drive.google.com/file/d/0B4vWHIIVWt9ncUUzcVl6VlI1ZVE/view?usp=sharing

So far I tried:
- recompiling it ("make clean" and "make") with DEFINES+=-DRGB_SLOWDOWN_GPIO=1 (also tried with 2), but they didn't go away
- the command with -P 2 and -P 3 but it also didn't seem to make any difference.

More hardware!  Sweet :) Assuming OSH Park is the best way to buy it?

Thanks again!

hzeller, 2015-10-22:
Mmmh, a few tips: make sure that you have solid ground connected everywhere - explicitly connect the ground from the LED panel inlets with the ground of your adapter circuit and with the ground of the Pi. It looks like the digital signal is not cleanly transferred always or has glitches.

Give each HCT125 a 100n capacitor across VCC/GND with shortest connection (I ususally just solder one accross the chip when used in breadboards).

Make sure that all the OE on the HCT125 are solidly on GND, otherwise their outputs float at high impedance.

Short cables.

Make sure the power supply is stable and possibly add a good capacitor to the direct inputs (the section about 'power' in the README).

And then, there are some LED panels, that actually have some ghost-pixels when others light up, which is hard to fix as it is a hardware problem there. You can typically see that with the ./text-example as it shows sharp letters on black ground: sometimes the is a 'halo' around letters.
(the halo you see around letters in the demo in the video are just simply because of antialiasing..)

hzeller, 2015-10-22:
OSH park is one way, I just put it there because it is easy to share, and easy to order at least when you are in North America. But the repository also contains the gerber files to send it to local fab house.

jmakacek, 2015-10-22:
Thanks for all the great info!  That explains it!

My wiring is a complete hackjob, wires are all 7+ inches long with multiples of them in between, my breadboard is some 10+ years old a friend gave me to play around with and my power supply is my MacGyver cut-off-electrical-tape-held-together-microusb-charger.  I basically got impatient waiting for parts and now that I look at it, I'm surprised it works at all.

Needless to say, PCB coming soon.

hzeller, 2015-10-22:
Yeah, that could explain things :)
Let me know how it works once you have put together the PCB.

jmakacek, 2015-10-22:
I have a few questions about the PCB(s) before ordering.

Mainly, for my current project I only need 1 led panel so should I get passive-rpi1 PCB (will it be fine with Pi2 or is there a Pi2 single panel version out there?), difference between active and passive, how they are powered, etc.  Basically, very noob questions.  And btw, I don't mind simply being pointed to docs or reading material, just trying to take a shortcut here :)

Is there a better place to ask (other than here) and is there a place where I can "contribute to your cause" somehow?

Thanks.

hzeller, 2015-10-22:
The Rpi1 version should work as well with the Pi2.
It doesn't have any active level shifting on it, so it only works with your panel if it can deal with 3.3V inputs (you typically see that if the input stage of your LED panel has HCT245 chips on there, then that will probably work fine; if they are HC245, then you most likely need level shifting)