hzeller/rpi-rgb-led-matrix #77: Segfault if used in openFrameworks

urshofer, 2015-10-28:
Hi -

I'm integrating the library in a openFrameworks addon. It should be a straight forward interface following the well known ``setup()  update()  draw()`` concept.

Anyway I'm stuck with a segmentation fault. Probably a threading issue:

```
Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0xb622d310 (LWP 6645)]
0x0014fe74 in rgb_matrix::internal::Framebuffer::DumpToMatrix(rgb_matrix::GPIO*) () at ../include/gpio.h:57
57          *gpio_clr_bits_ = value;
(gdb) bt
#0  0x0014fe74 in rgb_matrix::internal::Framebuffer::DumpToMatrix(rgb_matrix::GPIO*) () at ../include/gpio.h:57
#1  0x0014edf8 in rgb_matrix::RGBMatrix::UpdateThread::Run() () at led-matrix.cc:65
#2  0x001500f0 in rgb_matrix::Thread::PthreadCallRun(void*) ()
#3  0xb6f94c00 in start_thread () from /lib/arm-linux-gnueabihf/libpthread.so.0
#4  0xb67fe728 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
#5  0xb67fe728 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)
```

Has anybody an idea how to track down the bug? I'm a little bit stuck...

Best
Urs

hzeller, 2015-10-28:
Did you call Init() on the GPIO object ?

hzeller, 2015-10-28:
Looking at your repository at https://github.com/urshofer/ofxRpiLED/blob/master/src/ofxRpiLED.cpp#L21
you are allocating the GPIO object on the stack, then pass it to the RGB matrix, but then it falls out of scope.
You need to keep the GPIO object around for the life-time of the Matrix.

urshofer, 2015-10-28:
By the time writing you answered already my question :) You're quicker than my typing. I moved the GPIO object - problem solved. Now tackling others. Thanks!

closed 2015-10-28 by urshofer