hzeller/rpi-rgb-led-matrix #78: Only every 4 Row displayed

bradenb54, 2015-10-29:
Hi,

I have wired up my raspberry pi 2 following the new wiring diagram, I am using level shifters to shift the 3.3V logic to 5V logic. I'm having issues displaying all the rows. Only every 4 rows works (row 4,8,12,16) The rows that work seem to be displayed correctly. Do you have any suggestions for why this is working?

Thanks

hzeller, 2015-10-29:
Sounds like line A and B are not properly connected.

bradenb54, 2015-10-29:
They should be connected to pin 15 and 16 on the raspberry pi, right? when i disconnect them from the board(not connected from the matrix to anything), I get a fuzzy image in the pixels that were dead that resembles what its kind of supposed to look like.

hzeller, 2015-10-29:
So maybe something with the level shifters ?

bradenb54, 2015-10-29:
Ive tried different level shifters, and tested the level shifters with the other pins

hzeller, 2015-10-29:
You don't really give enough information to guess what is wrong in your set-up. What kind of panel you have, pictures of the wiring, description of your level shifters, parameters you use to the program etc.

hzeller, 2015-10-31:
Did you get it to work ?

closed 2015-11-12 by hzeller