hzeller/rpi-rgb-led-matrix #79: led image

rami2820, 2015-11-03:
Hi,

There was color, chains and output issues.
The color problem is solved. but I do not know hardly comes as a chain strangely.

Modify the source code is shown below.
lib/frame-internal.h 
source code:

union IoBits {
      struct {
        // This bitset reflects the GPIO mapping. The naming of the
        // pins of type 'p0_r1' means 'first parallel chain, red-bit one'
        //                                 GPIO Header-pos

        unsigned int unused_0_1     : 2;  //  0..1  (only on RPi 1, Revision 1)
        unsigned int p2_g1          : 1;  //  2 P1-03 (masks SDA when parallel=3)
        unsigned int p2_b1          : 1;  //  3 P1-05 (masks SCL when parallel=3)
        unsigned int strobe         : 1;  //  4 P1-07
        unsigned int p1_g1          : 1;  //  5 P1-29 (only on A+/B+/Pi2)
        unsigned int p1_b1          : 1;  //  6 P1-31 (only on A+/B+/Pi2)
        // TODO: be able to disable chain 0 for higher-pin RPis to gain SPI back.
        unsigned int p0_b1          : 1;  //  7 P1-26 (masks: SPI0_CE1)
        unsigned int p0_r2          : 1;  //  8 P1-24 (masks: SPI0_CE0)
        unsigned int p0_g2          : 1;  //  9 P1-21 (masks: SPI0_MISO
        unsigned int p0_b2          : 1;  // 10 P1-19 (masks: SPI0_MOSI)
        unsigned int p0_r1          : 1;  // 11 P1-23 (masks: SPI0_SCKL)

        unsigned int p1_r1          : 1;  // 12 P1-32 (only on A+/B+/Pi2)
        unsigned int p1_g2          : 1;  // 13 P1-33 (only on A+/B+/Pi2)
        unsigned int p2_r1          : 1;  // 14 P1-08 (masks TxD when parallel=3)
        unsigned int unused_15      : 1;  // 15 P1-10 (RxD) - kept free.
        unsigned int p2_g2          : 1;  // 16 P1-36 (only on A+/B+/Pi2)

        unsigned int clock          : 1;  // 17 P1-11

        unsigned int output_enable  : 1;  // 18 P1-12 (PWM pin: our timing)

        unsigned int p1_r2          : 1;  // 19 P1-35 (only on A+/B+/Pi2)
        unsigned int p1_b2          : 1;  // 20 P1-38 (only on A+/B+/Pi2)
        unsigned int p2_b2          : 1;  // 21 P1-40 (only on A+/B+/Pi2)

        unsigned int b              : 1;  // 23 P1-16
        unsigned int c              : 1;  // 22 P1-15  // row bits.
        unsigned int d              : 1;  // 24 P1-18
        unsigned int a              : 1;  // 25 P1-22

        unsigned int p2_r2          : 1;  // 26 P1-37 (only on A+/B+/Pi2)
        unsigned int p0_g1          : 1;  // 27 P1-13 (Not on RPi1, Rev1)
      } bits;
      uint32_t raw;
      IoBits() : raw(0) {}

The output image is 5       7 .

Run the command
> ./led-matrix -D1 -r32 -c2 message3.ppm

a demonstration: 
https://www.youtube.com/watch?v=GZl1-twGu98

Thank you.

rami2820, 2015-11-12:
Hi.

Still the question will be resolved afterwards.

The video is added to the above.

Run the command
> ./led-matrix -D0 -r32 -c2

a demonstration:
https://youtube.com/watch?v=zIRTWMb16i8

https://youtube.com/watch?v=e2pY6VV9ENc

I wait for answers. Thank you.

hzeller, 2015-11-12:
Looks like your second panel has a different pixel mapping, possibly an 1:8 multiplex. So it would be -r16 -c4 and then you need to write a transformer to map the pixels.

Just write a simple program that moves one pixel around, so that you get an idea how it is mapped internally.

rami2820, 2015-11-12:
When the output is too small to -r16 options.

 a demonstration:
   https://www.youtube.com/watch?v=avhN37bjAGc
and -r32 
a demonstration:
  https://www.youtube.com/watch?v=3gvbFpibP5M

We could understand that moves one pixel around using a simple program, But I'm not a Hardware developer, So Please I need more advice. Let us know how I can mapping that. I means the Way.

Thank you.

hzeller, 2015-11-12:
No hardware needed, at this part, it is just software.
It looks like the internal mapping in the panel is forward and back.

Set Pixels at various positions and figure out where they actually end up. Then write a transformer (include/transformer.h) that does the reverse mapping.

rami2820, 2015-11-12:
hm....Using simple ones by modifying Can you give me a sample code?

hzeller, 2015-11-12:
You really should first make an effort to understand the code yourself, or even look at it.
There are plenty of examples in lib/transformer.cc.

rami2820, 2015-11-12:
I will try to see in the lib / transformer.cc. Thank you.

closed 2015-11-12 by hzeller