hzeller/rpi-rgb-led-matrix #80: 3 parallel of 8 chains (64x32) each, only half is working

ayman0201, 2015-11-03:
Hi, 
I am following the ./led-matrix -D 0  with the 3 parallel active hub on a raspi 2
I am also using 12 64x32 panels ( connected as 3 parallel rows of 4 panels each)
When I provide -c 4 the animation shows perfectly on the first 2 columns of panels, if I increase it to 8 then only half shows and the other 2 columns of panels are either off or glitchy with no apparent image on them.
Any help? thanks

hzeller, 2015-11-12:
Sounds like a case for -DRGB_SLOWDOWN_GPIO (see README).

ayman0201, 2015-11-13:
I tried it on 1 and 2.
Results are a bit different:
assuming that we have 4(columns) x 3(rows) of panels and we have (0,0) as the top left one:
Only (1,1) and (2,1) are displaying. The top row is flickering and the bottom is off.
Any more suggestions?

ayman0201, 2015-11-13:
In other words: only second row is working

hzeller, 2015-11-13:
Then either your LED panels are broken or introduce too much timing imbalance (after all, this is now a shift-register that has to work over 256 bits), your active adapter is not soldered correctly (or only has HC chips not HCT; HC chips can't read 3.3V logic) or your power supply is not stable (make sure to have good, stable 5V on each panel, ideally with a beefy (> 1000uF), low-ESR capacitor close to each power input in particular if you have long cabeling.
Make sure that ground is well connected everywhere).

Given that two rows are off completely, I would first check the soldering of the active adapter and that you have the right logic chips with the "T".

ayman0201, 2015-11-13:
I do have the HCT245.
Without the GPIO slowdown I'm getting the right 2 columns working, when I enable it I'm only getting 2nd row to work. Any correlation?

Thanks

ayman0201, 2015-11-13:
Let me also add that the top row is glitchy, and the bottom row is completely off

hzeller, 2015-11-13:
Well, these things you have to debug yourself, this is nothing we can fix in software.

Your hardware is obviously flaky, so you have to fix that first. Please also read the README about various problems and how to fix these.

davemaster, 2015-11-14:
check the GROUND are common in all power supplies.

ayman0201, 2015-11-16:
Thanks for the feedback. The issue was in the cables. I had to swap 4 of them to make it fully work.

closed 2015-11-16 by ayman0201