hzeller/rpi-rgb-led-matrix #81: Artnet implemetation

alverman, 2015-11-10:
It 'the first time I write on GitHub and I hope not to create trouble.
I wanted to ask you a question. How could I implement the protocol artnet in your code?
I have a media server that sends the pixel data via artnet and I wish I could use your code to reconstruct the image output by the media server.
It would help if I could help to understand how to do.

Thank you very much, Alberto

Saij, 2015-11-10:
Hi.

As far as I unterstood you'll need to implement your own application/daemon.
In your own application you can use the library code provided in this repository to access the matrix.

Maybe there even is a Artnet library for python? That would make things much easier ^.^

Greetings
Saij

hzeller, 2015-11-10:
There is also an artnet bridge for PixelPusher, so you can use my PixelPusher implementation that uses this library: https://github.com/hzeller/rpi-matrix-pixelpusher (search for artnet)

closed 2015-11-11 by hzeller