hzeller/rpi-rgb-led-matrix #82: Can't 'make'  led-image-viewer

natemac, 2015-11-12:
Whenever I try to compile the led-image-viewer I get the following error

pi@raspberrypi ~/rpi-rgb-led-matrix $ sudo aptitude install libgraphicsmagick++1-dev
No packages will be installed, upgraded, or removed.
0 packages upgraded, 0 newly installed, 0 to remove and 47 not upgraded.
Need to get 0 B of archives. After unpacking 0 B will be used.

pi@raspberrypi ~/rpi-rgb-led-matrix $ make led-image-viewer
g++ -Iinclude -Wall -O3 -g `GraphicsMagick++-config --cppflags --cxxflags` -c -o led-image-viewer.o led-image-viewer.cc
make -C lib
make[1]: Entering directory '/home/pi/rpi-rgb-led-matrix/lib'
make[1]: Leaving directory '/home/pi/rpi-rgb-led-matrix/lib'
g++ -Wall -O3 -g led-image-viewer.o -o led-image-viewer -Llib -lrgbmatrix -lrt -lm -lpthread `GraphicsMagick++-config --ldflags --libs`
/usr/bin/ld: cannot find -lwebp
collect2: error: ld returned 1 exit status
Makefile:35: recipe for target 'led-image-viewer' failed
make: *** [led-image-viewer] Error 1

hzeller, 2015-11-12:
did you try to install libwebp-dev ?

natemac, 2015-11-12:
That fixed everything. Thank you

closed 2015-11-12 by natemac