hzeller/rpi-rgb-led-matrix #84: led-image-viewer with Adafruit HAT

natemac, 2015-11-12:
When I used the pin jumpers to the 16x32 I was able to run the Image Viewer with no issue, but I was limited to the amount of panels I could use.
I added the Adafruit HAT to recompiled the code with DEFINES+=-DADAFRUIT_RGBMATRIX_HAT
The demos work fine, but when I use the Image Viewer now, it no longer works.
Any thought of a solution? Thanks for all the help the last day or so.

hzeller, 2015-11-12:
mmh, so the image viewer should not be different than the demos.

Always a good idea to do a `make clean` and re-build the binaries in case a flag hasn't been picked up correctly.

natemac, 2015-11-12:
doing that as we speak, dumped everything and started from scratch
will update you, thank you for everything

natemac, 2015-11-12:
seem to do the fix, guess i should take the advice I give my aunt "did you try rebooting yet?"

closed 2015-11-12 by natemac

hzeller, 2015-11-12:
:)
Have fun with your new image viewer device.