hzeller/rpi-rgb-led-matrix #86: I have no idea whats going on here...

lifelikely, 2015-11-17:
Have a single 16X32 matrix connected to Raspi 2, using the wiring diagram you detail. (i referenced the happy face emoticons because I am just using one matrix, not daisy chaining)

 I run "sudo ./led-matrix -r 16 -D 5" and this is the result... what could possibly be wrong here? I did the wiring twice and got the same result. Pi is not overclocked or anything.

https://youtu.be/Mddz_1MVMYw

davemaster, 2015-11-18:
Check Raspberry Pi pinout and cabling AGAIN....

hzeller, 2015-11-18:
* Verify that the chips close to the input stage at the display are HCT245 or AHCT245. If they are only HC245, they can't deal with the 3.3V  logic levels from the Pi (reduce the voltage to the panels to 4V or so to see if that is better).
 * Make sure to have GND properly connected everywhere.

Read the following paragraph in the README, it contains some useful tips:
https://github.com/hzeller/rpi-rgb-led-matrix#help-some-pixels-are-not-displayed-properly

lifelikely, 2015-11-18:
Yup.... not the wiring davemaster. I verified MANY times. I think the problem is that the chip hzeller refers to is an HCT245. Got to shift the logic. hzeller: do you mean bring down the power supply running the matrix to 4v? I thought the matrix requires 5v to run? Retailer said that I might need to shift the logic from the Pi.
thanks for the feedback,. really helps.

hzeller, 2015-11-18:
If it is a HCT245 it should not have problems with the 3.3V logic (only the HC245 has). In that case bringing the power of the panel down to 4V might verify that things start working (but the colors will be off, it is not a permanent solution).

Solid ground connection between Pi and panel GND and a power supply that can support the ~2A needed for this display are important. Also the power supply should have good transient response which means you need short, thick cables and possibly a low-ESR capacitor right at the panel power entrance.

Your output looks very erratic, it should at least show something. Try the `-D0` or the `-D3` examples to see if this yields something recognizable.
Did you go through the checklist in the link I sent ? In particular the slowdown option  might be useful.

lifelikely, 2015-11-18:
the full label on the chip ICN74HC240TS, I think.

[image]

hzeller, 2015-11-18:
Yeah, the HC245 do have trouble with the logic level. You might need to solder up an active adapter
https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3

lifelikely, 2015-11-18:
Thank you for sharing all the info!!

This is being used in an MFA project, so it's important that the results are predictable. Colors need to be spot-on. So taking the matrix' power down to 4v doesn't sound like a good idea. 

I just don't have the resources to print your board, which looks very cool btw. I was hoping I would be able to use the 74HCT245 chip I have, but I am clueless as to how to wire it in conjunction with the Pi and matrix. I have capacitors and resisters, so I think I could make the circuit, just absolutely blank about where everything goes.

hzeller, 2015-11-19:
Here is the schematic: https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/adapter/active-3/schematic.pdf

You essentially have to wire the inputs to one side, the outputs to the other side and set the right direction. If you look at the datasheet of a 74HCT245 you see what I mean (it can go both directions).

If operated at 5V, it outputs 5V signals, but happily can accept and understand 3.3V logic levels on the inputs.

You could as well also get two HCT245 in the size they are on the LED panel and replace these (only the first panel in the chain is important). But desoldering and re-soldering might be a bit harder.

lifelikely, 2015-11-20:
got it working. without needing to shift logic. Adafruit admin told me to stop using the library I was using, and use the one found here: github.com/hzeller/rpi-rgb-led-matrix/archive/master.zip

And everything worked!! 

Thank you very much for helping out. You are quite the gentleman.

cheers

hzeller, 2015-11-20:
Errr, yes, this is my library. And notice the URL above, you filed this issue in the context of this library.

So, you were using the ancient Adafruit fork all along ? Also, they have some ancient tutorials that are not really properly reflecting current reality.

lifelikely, 2015-11-20:
I totally agree that this whole business would not have gone down if they only offered updated tutorials for perpetual noobs like me. 

Yeah, I just downloaded as prescribed in the tutorial (git clone https://)github.com/hzeller/rpi-rgb-led-matrix/ and was working from that. :-/

hzeller, 2015-11-21:
Alright, so we can close this now :)

closed 2015-11-21 by hzeller