hzeller/rpi-rgb-led-matrix #87: led-image-viewer compile error /usr/bin/ld: cannot find -lwebp

jwheare, 2015-11-19:
I'm getting an error trying to `make led-image-viewer`

`/usr/bin/ld: cannot find -lwebp`

Here's the full log and info on my installed version of `libgraphicsmagick++1-dev`

Are there other deps I should install?

(Also I noticed the source comment still says to install image-magick even though the README has been updated to say GraphicsMagick https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/led-image-viewer.cc#L17)

```
pi@raspberrypi ~/led/rpi-rgb-led-matrix $ make
g++ -Iinclude -Wall -O3 -g -c -o demo-main.o demo-main.cc
make -C lib
make[1]: Entering directory '/home/pi/led/rpi-rgb-led-matrix/lib'
g++ -I../include -Wall -O3 -g -fPIC -DRGB_SLOWDOWN_GPIO=1 -DADAFRUIT_RGBMATRIX_HAT -c -o gpio.o gpio.cc
g++ -I../include -Wall -O3 -g -fPIC -DRGB_SLOWDOWN_GPIO=1 -DADAFRUIT_RGBMATRIX_HAT -c -o led-matrix.o led-matrix.cc
g++ -I../include -Wall -O3 -g -fPIC -DRGB_SLOWDOWN_GPIO=1 -DADAFRUIT_RGBMATRIX_HAT -c -o framebuffer.o framebuffer.cc
g++ -I../include -Wall -O3 -g -fPIC -DRGB_SLOWDOWN_GPIO=1 -DADAFRUIT_RGBMATRIX_HAT -c -o thread.o thread.cc
g++ -I../include -Wall -O3 -g -fPIC -DRGB_SLOWDOWN_GPIO=1 -DADAFRUIT_RGBMATRIX_HAT -c -o bdf-font.o bdf-font.cc
g++ -I../include -Wall -O3 -g -fPIC -DRGB_SLOWDOWN_GPIO=1 -DADAFRUIT_RGBMATRIX_HAT -c -o graphics.o graphics.cc
g++ -I../include -Wall -O3 -g -fPIC -DRGB_SLOWDOWN_GPIO=1 -DADAFRUIT_RGBMATRIX_HAT -c -o transformer.o transformer.cc
ar rcs librgbmatrix.a gpio.o led-matrix.o framebuffer.o thread.o bdf-font.o graphics.o transformer.o
make[1]: Leaving directory '/home/pi/led/rpi-rgb-led-matrix/lib'
g++ -Wall -O3 -g demo-main.o -o led-matrix -Llib -lrgbmatrix -lrt -lm -lpthread
g++ -Iinclude -Wall -O3 -g -c -o minimal-example.o minimal-example.cc
g++ -Wall -O3 -g minimal-example.o -o minimal-example -Llib -lrgbmatrix -lrt -lm -lpthread
g++ -Iinclude -Wall -O3 -g -c -o text-example.o text-example.cc
g++ -Wall -O3 -g text-example.o -o text-example -Llib -lrgbmatrix -lrt -lm -lpthread

pi@raspberrypi ~/led/rpi-rgb-led-matrix $ make led-image-viewer
g++ -Iinclude -Wall -O3 -g `GraphicsMagick++-config --cppflags --cxxflags` -c -o led-image-viewer.o led-image-viewer.cc
make -C lib
make[1]: Entering directory '/home/pi/led/rpi-rgb-led-matrix/lib'
make[1]: Leaving directory '/home/pi/led/rpi-rgb-led-matrix/lib'
g++ -Wall -O3 -g led-image-viewer.o -o led-image-viewer -Llib -lrgbmatrix -lrt -lm -lpthread `GraphicsMagick++-config --ldflags --libs`
/usr/bin/ld: cannot find -lwebp
collect2: error: ld returned 1 exit status
Makefile:35: recipe for target 'led-image-viewer' failed
make: *** [led-image-viewer] Error 1

pi@raspberrypi ~/led/rpi-rgb-led-matrix $ aptitude show libgraphicsmagick++1-dev
Package: libgraphicsmagick++1-dev 
State: installed
Automatically installed: no
Version: 1.3.20-3+rpi1
Priority: optional
Section: libdevel
Maintainer: Laszlo Boszormenyi (GCS) <gcs@debian.org>
Architecture: armhf
Uncompressed Size: 1,457 k
Depends: libgraphicsmagick++3 (= 1.3.20-3+rpi1), libgraphicsmagick1-dev
Suggests: graphicsmagick
Conflicts: libgraphicsmagick++-dev
Replaces: libgraphicsmagick++-dev
Provides: libgraphicsmagick++-dev
Description: format-independent image processing - C++ development files
 GraphicsMagick provides libraries in several programming languages to read, write and manipulate image files across a large number of
 formats, from the widely used jpeg, tiff, bmp or xpm to special-purpose formats such as fits or image formats found on some photo CDs.
 There are functions for finegrained image processing tasks, as well as conversion routines between the various image formats. 
 
 The GraphicsMagick library is a fork of ImageMagick and therefore offers an interface that is similar in features, but intended to be
 more stable across releases. While compatibility does not go so far that the GraphicsMagick library serves as a drop-in replacement
 for ImageMagick, conversion can usually be done with little effort. 
 
 This package contains the C++ development headers and library files needed to compile programs using the GraphicsMagick++ library.
Homepage: http://www.graphicsmagick.org/

pi@raspberrypi ~/led/rpi-rgb-led-matrix $ 
```

jwheare, 2015-11-19:
Installed `libwebp-dev:armhf (0.4.1-1.2)` and it now builds but warns about `/usr/bin/ld: warning: libjpeg.so.8, needed by /usr/lib/libGraphicsMagick.so, may conflict with libjpeg.so.62`

Hopefully this can help anyone else who had this issue.

closed 2015-11-19 by jwheare

jwheare, 2015-11-19:
And completely missed that this is a dupe of #82 sorry!

crale, 2016-09-13:
I too am getting "/usr/bin/ld: warning: libjpeg.so.8, needed by /usr/lib/libGraphicsMagick.so, may conflict with libjpeg.so.62" even though I installed all the extra libraries. How can I fix this?

hzeller, 2016-09-13:
Did you look at the Readme that describes what packages to install?

crale, 2016-09-13:
Yes, I installed all of them.

hzeller, 2016-09-13:
ok. the libjpeg warning you can ignore.

crale, 2016-09-13:
Ok, it compiles, yes...but when I run it and pass a gif file to display, it does not display it. I also don't know how to exit the image viewer. I end up having to close the terminal and then open it again.

I just use the default settings since I am trying with a 32x32 panel.

hzeller, 2016-09-13:
The usual Ctrl-C would be to stop the image viewer. If you just closed the terminal, chances are that the old image viewer is still running, attempting to access the same hardware. Make sure to kill all instances (or simply reboot).

The other demos are running? Did you read the Troubleshooting section? Did you disable sound?

crale, 2016-09-13:
hey, thanks. I forgot to add HARDWARE_DESC=adafruit-hat

It is working now (a little flickering though). I really appreciate the help.

hzeller, 2016-09-13:
the flickering you can fix with the hardware mod for the Adafruit HAT.