hzeller/rpi-rgb-led-matrix #91: List of commands in the python bindings?

billthefighter, 2015-11-26:
Hello,

I recently upgraded the firmware in my project to run the latest version of this code instead of the older, less stable, much more flickery adafruit code. However, it appears that adafruit did a lot of work to incorporate bibliopixel and some other libraries that trivialize some of what I was doing. 

My project is very simple, but it would be greatly streamlined if I had a list of the available python commands I could use to manipulate the RGB matrix. Is there such a list? I am still a very junior programmer and as such this may be an obvious question - if so any assistance would be greatly appreciated.

hzeller, 2015-11-26:
In general, the Python API wraps the available C++ Api, so it provides you essentially a frame-buffer to write to. You see a couple of examples in the python/samples directory.

The important thing is the class `RGBMatrix` which provides you way to SetPixel() or Clear(). With that, you can easily wire it up to other libraries, e.g. bibliopixel (I don't know that library, but it sounds sufficiently simple). I think a very similar class in the Adafruit code is called `Adafruit_RGBmatrix`, so you might be able to make the transition.

You might also be interested in trying the pixel-pusher protocol ( https://github.com/hzeller/rpi-matrix-pixelpusher ), and then code animations using the processing framework ( https://processing.org/ ) which allows to program all kinds of interesting hacks.

@Saij contributed the Python API, so he might be able to give you a better answer than me regarding the Python API and a list of functions.

As you said @adafruit already has a set of Python libraries that specifically integrate all kinds of things to make pixel programming simple, so ideally I'd wish them to use the RGBMatrix here (and possibly send pull requests for improvements if needed). They have used a fork of my library a while ago, added the Python library and then essentially abandoned - maybe we can motivate them to refresh the Python interfacing with the current version of the code ?

billthefighter, 2015-11-28:
Thanks for the feedback - I've contacted adafruit to see if they are planning to get cracking on an updated implementation. If not, I will try to learn how to wire bibliopixel or pixel-pusher into your library myself as a good exercise. 

@Saij - do you have any advice on a logical direction to proceed?

hzeller, 2015-12-02:
BTW, I added two examples `graphics.py` and `runtext.py` in the `python/samples` directory to help get you started with all the graphics functions that are also available in the library (the Python library is a wrapper around the C++ lib, so it is essentially what is provided there).

Saij, 2015-12-02:
As @hzeller already said, the python library is just a wrapper around the C++ code.
I tried to made it as close as possible to resemble the C++ interface.

The best sample to start ist https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/python/samples/simple-square.py
There you just set some pixels and use the backbuffer feature.

For more graphics (like fonts, lines and circles) just take a look at https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/python/samples/graphics.py

For setting up the matrix with python take a look at https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/python/samples/samplebase.py
There you have the basic setup (creating the object, initializing all and so on)

liquidthex, 2016-01-10:
Hello thank you for your wonderful library.
I am trying to use the Python bindings for a custom clock project.

With the adafruit python fork they added a method called SetImage, this allowed you to pass it an image id from ImageMagick.

Is there anything like that in the current libraries or would I essentially have to implement the SetImage method?

Link to SetImage method in adafruit fork:
https://github.com/adafruit/rpi-rgb-led-matrix/blob/master/rgbmatrix.cc#L144

hzeller, 2016-01-10:
Currently, there is no SetImage, but I accept pull requests :)
I think it would be useful for the Python library to have a SetImage() to improve performance. When you implement that, also consult with @Saij who did the initial implementation of the Python API.

liquidthex, 2016-01-11:
To be honest I don't know how I would incorporate it into the library, but here's what I came up with for my project:
```python
def setimage(im,mx):
  (w,h) = im.size
  pix = im.load()
  for x in range(w):
    for y in range(h):
      (r, g, b) = pix[x, y]
      mx.SetPixel(x, y, r, g, b)
```
im = Image object from Image.new()
mx = RGBMatrix object from RGBMatrix()

Seems to work decent :)

Saij, 2016-01-11:
I think that this is the correct way ^.^
Only other I can image is directly in Cython.

maddogs, 2016-01-16:
@liquidthex did you use the setimage method using Henner's library or Adafruits? If Henners, do you have a sample you can provide?

liquidthex, 2016-01-16:
@maddogs I am using Henner's library due to Adafruit's being out of date. Unfortunately Henner's Library does not have a setimage method. I created my own setimage method in python; scroll up 3 comments to see the code.
Here's the project I'm working on that I used it in: https://github.com/liquidthex/ThexBerryClock/blob/master/ThexBerryClock.py#L225
(Warning: Spaghetti code)

maddogs, 2016-01-17:
@liquidthex Excellent, thank you. Got it working.

maddogs, 2016-01-17:
Using the existing sample code and @liquidthex  setimage() definition, I've put together a basic sample for displaying basic png \ jpg image files using the python library

https://github.com/maddogs/RGB-LED-Matrix-Images

I've only just started to learn, so any suggestions, corrections are welcome :)

techtravis, 2016-05-18:
I modified the code above to handle scrolling the image from the right to left.
But it doesn't appear to be the most efficient as the image scroll speed slows down depending on the length of the image you are trying to scroll.

Any help on how to speed this up?

```python
	def setimage(im,mx):
		(w,h) = im.size
		pix = im.load()
		offscreenCanvas = mx.CreateFrameCanvas()
		pos = offscreenCanvas.width
		while (pos + w > 0):
			pos -= 1
			for x in range(w):
				for y in range(h):
					(r, g, b) = pix[x, y]
					offscreenCanvas.SetPixel(x+pos,y,r,g,b)

			time.sleep(0.015)
			offscreenCanvas = mx.SwapOnVSync(offscreenCanvas)
```

ethur, 2016-06-12:
I was in a similar boat, wanting to use Adafruit's older "rgbmatrix.cc" Python library with the current RGBMatrix library, and I documented my solution on Adafruit's Forum:
[https://forums.adafruit.com/viewtopic.php?f=50&t=98006](url)

The update was fairly simple -- their library needed the third "parallel" argument added, and a function involving "SetWriteCycles" had to be remarked out. It's still early to tell if this is all that was required, but it allowed me to get my older Python scripts using SetImage (and all the other wonderful PIL functions, such as ImageFont) to work with the current **hzeller** library.

FarhadGSRX, 2016-06-15:
@ethur FYI, the link to the forum you posted is not currently working. Thank you for your explanation. I will be using it soon!

ethur, 2016-06-15:
Thanks for the note -- it was my first time posting on GitHub. I've corrected the link.

FarhadGSRX, 2016-07-08:
Hey @ethur! So I finally got around to running through your instructions.

After running through your instructions twice and checking for mistakes, I don't see the SetImage() and other functions becoming available in the RGBMatrix class like I expected...

The functions are supposed to show up there, with the other functions, right? (e.g. Clear, CreateFrameCanvas, Fill, SetPixel, etc)

Or are the functions available elsewhere? Please let me know if you need more info from me to help. Thanks in advance!

closed 2017-07-30 by hzeller