hzeller/rpi-rgb-led-matrix #92: Lit & Faint pixles

maddogs, 2015-11-27:
Hi - Any suggestions on what is causing the faint lit leds on this matrix?

[image]

I'm using a Pi2, Adafruit Hat and just the sample ./text-example function.

I've tried different fonts, changed the HAT cable, changed the GPIO speeds under Makefile, using Clean make as the command. (The makefile is set to use the HAT)

I've also tried changing the x \ y positions etc.

These weird pixels only appear using the text function. The other demos are fine. This happens if only one 16x32 panel is connected.

The only thing I haven't tried is directly wiring the Pi to the Panels.... I didn't have enough jumper cables, they are on order!

If it's a 'thing' with the text function, no problems :)

Cheers Steve

hzeller, 2015-11-28:
The shadow pixels are not sent to the panel, it is the hardware of the panel not cleanly shutting off pixels. So it is mostly a quality issues with some LED panels.

It typically shows with very high contrast situations like in the text examples.

Maybe it is possible to find a workaround in software to trick them to behave correctly (by tweaking some timings); as soon as I find such a misbehaving panel, I'll poke at it with a scope to see if there is something to be done.

You could try playing with timings, e.g. set DEFINES+=-DRGB_SLOWDOWN_GPIO=1 to 2 instead (in lib/Makefile), or set the kBaseTimeNanos in lib/framebuffer.cc to something like 200 or 300 and report back if that makes any difference.

But until I have seen this behavior with a scope and maybe can work-around it, this is a little artifact to live with :/

maddogs, 2015-11-28:
Hi - Thanks, I've changed the kBaseTimeNanos to 300 and the pixels do not appear to be so bright :)

It doesn't cause any issues at all and your software works great. Thanks for you time!

Steve

closed 2015-11-28 by maddogs