hzeller/rpi-rgb-led-matrix #93: LED 16x32, Raspberry PI1, problem with G2

mschoett, 2015-11-29:
Hi, 

I am using an Adafruit 16x32 LED matrix with a Raspberry PI1 with new wiring according to the description. Everything works fine except the green colour is only showing up on the bottom 8 lines when running demo 4  - Pulsing color. (Green at the top 8 lines also missing for the other demos).

I switched G1 & G2 on the display an then I see green on the top 8 lines (but not on the lower 8), so at least the display seems to be fine. 

I checked my wiring of G2 going into pin 21 on the Raspberry PI which is correct according to the description. And in the Makefile I have of course NOT switched to DRGB_CLASSIC_PINOUT. 

Any ideas? May be there is a problem with the Pi? Is there some test program for the Pins?

Thank you,

Michael

hzeller, 2015-11-29:
Your Pi has probably a broken GPIO pin, yes.

You could probably find a free GPIO pin and then remap the p0_g2 to somewhere else:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/framebuffer-internal.h#L186

Just rename the current p0_g2  variable to unused_9 and look for a GPIO that is unused and accessible in your Pi1 on the limited. For instance, you could put it where there is now p2_r1 (GPIO 14, Pin 8 on the header).

mschoett, 2015-11-30:
Thank you very much for your fast reply. I just did your suggested remapping but it did not help. However, I then checked my IO-Ports using a breadboard and an LED and found out that one of the connector cables was broken :-(

Everything is working fine now. I am sorry for bothering you.

hzeller, 2015-12-02:
Cool. Sometimes it is these simple electrical issues :)

closed 2015-12-02 by hzeller