hzeller/rpi-rgb-led-matrix #96: Support for 16x64 displays

lanrat, 2015-12-01:
Moving the discussion from pull #88 here.

What is the purpose of `double_rows_(rows / 2)` in framebuffer.cc (https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/framebuffer.cc#L55)?
When changing to `double_rows_(rows)` my display works perfectly with `rows=16` and `chain=2`.
Should this be made into a variable that can be controlled in the API?

lanrat, 2015-12-02:
You can view the change I made to work on my display here: https://github.com/lanrat/rpi-rgb-led-matrix/commit/9c11e9937c233da33928e9ee72e9a0ef3e956400

I think that this should be user-configurable and not hard-coded, but I don't understand it well enough to be sure.

hzeller, 2015-12-03:
Replacing the rows/2 with rows would work in your case. I need to think about a way to configure that that the documentation is still understandable :)

In general, this is just a special case of only using half  a 32 high display (as you already have seen in the pictures of your pull request).

Which means - did you notice that with that display, you have the opportunity to connect up to 6 chains ? Usually, one display gets both R1G1B1 and R2G2B2. Now, you connect the first to the first display, the second to the next display. So each previous single chain can be split in two.

So, if you have a split chain with two 64x16 displays, (first connected to R1G1B1, second to R2G2B2, all the other signals A,B,C,D, Strobe, Clk etc connected to both), you can say rows=32 and chain=2 and essentially get a 64x32 display.

If you go with -P3, you can have up to 64x96 pixels (and of course, each chain can be longer, so you can have multiple of 64 in x-axis direction).

Do you have a close-up picture of the connector on your board ? (similar to https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/img/hub75-other.jpg ). Hopefully, it has some readable description. I'd like to add that case to the documentation.

lanrat, 2015-12-03:
Here is a photo of the connector on my display
[image]

I only have 1 display which I bought off [ebay](http://www.ebay.com/itm/141637118333) with no documentation, so I had no idea how it works, let alone the amount of displays that can be chained.
I'm also using a Raspberry Pi B so I have no parallel support.

hzeller, 2015-12-03:
Thanks for the picture - it looks like it is pretty different from the 'usual' connector, yet the electric stuff seems to work similarly.
Anyway, thanks to  each display only  'half' a 32 display, you now have parallel support for two chains :)

commit a9345e18 (2015-12-05): o Make it configurable to only have one sub-panel per panel, as some
  older panels provide that. Fixes #96
closed 2015-12-05 by hzeller