hzeller/rpi-rgb-led-matrix #97: Python graphics module does not function

lanrat, 2015-12-01:
When testing the python bindings everything defined in [core.pyx](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/python/rgbmatrix/core.pyx) works great, but when using any of the functions from [graphics.pyx](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/python/rgbmatrix/graphics.pyx) nothing is displayed.

Below is a sample program that I think should draw a vertical red line, however it does not print anything on the display. Manually calling `SetPixel` works fine.

```
#!/usr/bin/env python
from rgbmatrix import RGBMatrix
from rgbmatrix import graphics
import time

rows = 16
chains = 2
parallel = 1
matrix = RGBMatrix(rows, chains, parallel)

red =  graphics.Color(255, 255, 255)
graphics.DrawLine(matrix, 10, 10, 0, 10, red)

time.sleep(10)
```

hzeller, 2015-12-01:
@Saij can you have a look ?

hzeller, 2015-12-02:
graphics.Color is a class that has red, green and blue properties, but no constructor with these values. It should probably have one. Will add it.

hzeller, 2015-12-02:
Fixed in https://github.com/hzeller/rpi-rgb-led-matrix/commit/cd0a5e3fbfe25163907910f1a58a304d390e275a

@Saij can you have a look if I did not screw up Python ?
Generated code with cython 0.23.4

closed 2015-12-02 by hzeller

Saij, 2015-12-02:
Looks fine. Thanks for fixing it/adding the constructor