hzeller/rpi-rgb-led-matrix #101: Make transformers available from Python

hzeller, 2015-12-04:
Transformers should be accessible by name from Python.

bentwire, 2016-01-01:
I'm new to Cpython, so I'm not sure, is this a pretty straight forward process?  From what I see in the core and graphics python bindings it looks simple enough...

daker, 2016-01-06:
👍  for this

yarikoptic, 2016-01-22:
+100 on this one ;)
I would really appreciate if someone has implemented this since I would like to script display from Python but need to establish a chain of transformers (e.g. https://github.com/hzeller/rpi-rgb-led-matrix/pull/113 ;-) ) for my 2x2 display.

Saij, 2016-02-16:
If I'll get my testing box up and running again I'll implement it ^.^
(But first get my box back again from my girlfriend -.-)

daker, 2016-02-18:
@Saij We will be grateful for that :+1:

delectatronics, 2016-04-04:
Hi Saij
Do you have an estimate when you believe the transformers will be available from Python?  I am trying to have a baseball scoreboard operational using Python 2.7 in about 3 weeks.
I appreciate your time and I am not pushing at all - just trying to work out if I need to go in a different direction.
Cheers

Saij, 2016-04-04:
Sorry just got back from a convention ^.^
Will give my best to get it running soon ^.^ But can't promise anything yet

delectatronics, 2016-04-05:
No probs at all – I appreciate all of your voluntary efforts

serverfire, 2016-08-30:
@Saij this should open , many need transformer for python , thank you

hzeller, 2016-08-30:
I am working towards it

commit 9f98e15b (2018-03-17): Provide pixel_mapper_config option and --led-pixel-mapper flag in Python.

Fixes #101
closed 2018-03-17 by hzeller