hzeller/rpi-rgb-led-matrix #99: Is it possible to use this library with HUB12 single colour panels?

Jobbies, 2015-12-04:


hzeller, 2015-12-04:
I don't know, I don't have one. Do you have a datasheet ?
If they clock in/strobe data similarly and possibly have a concept of clocking in data (some internet search seems to confirm that), it should be possible to make it work (though the pinout is probably different, you have to go by signal name).

Jobbies, 2015-12-04:
Here is a schematic of the pins on the connector. 
https://cdn.shopify.com/s/files/1/0045/8932/files/DMDCON_DMDConnector.pdf

The pins seem similar there is only one R pin used.
GND, nOE, A, B, CLK, SCLK, R

http://forum.freetronics.com/viewtopic.php?t=5580

Jobbies, 2015-12-04:
It's hard to find a datasheet for these, but I did find this.
[P10sch.pdf](https://github.com/hzeller/rpi-rgb-led-matrix/files/51809/P10sch.pdf)

hzeller, 2015-12-04:
Yep, looks like you can make it to work. Just connect one color to the color input (R), CLK to CLK, SCLK to latch and A, B to A, B.

Bonus: Since this display only uses one color, but this code outputs six colors in parallel (RGB top and bottom), you can connect up to six parallel displays of this kind to a single chain, or up to 18 displays for what would be three RGB chains (`-P3`) (but you need to map the different chains to differerent colors by implementing a Transformer class).

I don't have such displays available to test, but it doesn't look like much work to connect.

Jobbies, 2015-12-05:
I had hoped it would be that simple. When I wire it up like that and run any of the examples all of the LEDs light up. Then when the examples are finished every 4th row of the matrix stays lit.

https://flic.kr/p/BicDaB

hzeller, 2015-12-05:
Since you only have A, B, you need to give the `-r 8` option to get the 1:4 multiplexing. I suspect that you got multiple rows multiplexed into each other so that they essentially ended up looking all lit.

Also important: logic levels: If the chips on your display are of the HC type instead of HCT (e.g. 74HC245 instead of 74HCT245), you need to operate the panel closer to 3.3V than 5V so that the logic levels are properly intepreted (The Raspberry Pi logic level is 3.3V). Or add a HCT245 bus driver operated at 5V to do the logic level shifting (have a look at the `adapter/` directory to get an inspiration).

The residual lit row might be because output enable is not pulled to high in the end ? You might want to have a 10k resistor from the Pi output enable to 3.3V.

hzeller, 2015-12-05:
Looks like your panel is wrapping around multiple times, to fill the area, so it probably looks like a panel with (32*4) pixel long and 4 pixel high (since this doesn't have two half-displays, it is only 4 pixels instead of 8 pixels. So with `-r8` the library would treat it as a display that is 8 pixels high, but only the first four are displayed.

For that, I now added an option in the `lib/Makefile` that you can add: `-DONLY_SINGLE_SUB_PANEL`. That allows easier reasonable mapping.

You still would need to write a mapper that maps the stretched display as it presents itself into the mapping it actually shows (so column 0..31 are the same, column 32..63 are actually column 0..31 on Y+4 and so on).

Also, have a look at Bug #96 - that user has a yet different display, but there are similarities to yours.

koderov, 2016-05-06:
hello, I too bought some white panels, 32x16 , chip sm74hc595a , I get the 4th row remaining lit.What is this mapper you are talking about , and where is it included,
My panel is a hub12 with pins
 OE
F(Dont know*)
A,B,
N(Grnd),
S(strobe),
L(latch),
R(red)

kapiau, 2016-11-16:
hello,
I'm too am getting a handful of 32x16 modules, chain them to a HUB12 and use as a display.
Has anyone fully accomplished such task in a way that you could share a "getting started" guide?
Even if all you have is a low level description, I very much welcome any input.

akhiiasati, 2018-09-28:
I want to interface hub12 with 8051 microcontroller.can anybody will help

closed 2020-05-07 by Jobbies