hzeller/rpi-rgb-led-matrix #104: Static scanning?

AnjinMeili, 2015-12-09:


hzeller, 2015-12-09:
I don't have any of the boards and you didn't much describe how they are wired up.
But if you merely need 1:1 scanning and no parallel input, you can
  a)  remove the rows-test (rows != 8 && rows != 16 && rows != 32...) somewhere in demo-main.
  b) uncomment #DEFINES+=-DONLY_SINGLE_SUB_PANEL in lib/Makefile
then depending on how your display is wired up, you'd want to write a transformer (include/canvas.h, examples include/transformer.h) to do some useful mapping that maps the physical space (typically wrapped around multiple times) to a logical thing.

If these panels are single-color, you can as well clock in up to 6 in parallel in with a single output (if you    use 'red1' for the first rows, 'green1', second  ... 'blue2' sixth) - you have to map these colors to the underlying Canvas though. So if you can access the 6 rows individually in your panel, then you can clock them in in parallel (but need to have some row to color mapping in the transformer); if they are internally chained, then you need to unmap these from column to row.

If you use all parallel chains for color, that means in total 3 * 6 = 18 panels in parallel  in one color.

Also have a look at #96 and #99 - these users have apparently similar displays, but since I don't have any like these I only can hint you guys in the right direction to change the code. It sounds very doable and it should not take more than an hour to get it to work.

AnjinMeili, 2015-12-09:
Pins are similar to HUB40.  The ones I am toying with the most have 

      R G B CLK LE OE/R OE/G OE/B  

for each top and bottom 8x16 piece of board.  

They are RGB, designed for daylight use.  

Global brightness per color can be adjusted with PWM on the OE pins.

Couple simple examples

https://github.com/AnjinMeili/1616-RGB-LED-Matrix

I will  review the above thou and play some more, thanks for the aid.

hzeller, 2015-12-09:
So if this is 1:1, then you can probably make this light by connecting all the OE together (are they positive or negative logic ? The library assumes negative logic), have CLK connect to CLK and LE (which I assume to be 'latch') connect with strobe. Connect the both RGB to the top and bottom part of the board.

if the board is 8x16, then this is 128 colors to clock in; the library assumes 32 per panel, so from the perspective of the library, these are 4 chained panels. So the rough options would be to
  `-r 2 -c 4`  (chain of four with 'two lines', which is the long 128 pixel top line and bottom line).

then you need of course write a transformer that converts the positions.

AnjinMeili, 2015-12-09:
Positive, with ability to PWM the OE pins for global brightness.

Board is 16x16, but split into two 8x16 paths.

Will try above.

hzeller, 2015-12-09:
The library actually will use the OE to display a grayscale image with 24 Bit color!

AnjinMeili, 2015-12-10:
Well, panel lights.  But ordering is off a lot.  I ended up daisy chaining 2 16x16 panels.  I figure this should make an object similar to a 16x32.  

I modified as suggested, but when I try -r2 -c4, PI reboots.  Other modes provide sporadic results.

Tried just one board as well... I need a bigger power supply for sure.  Seems these outdoor panels can suck down a good 6 amps.  

And... I am sure one of these insanely bright panels could find its way to you if your curiosity is high enough to wish to dig deeper.

I am using a Adafruit RGB Matrix Hat btw.  I made a harness today for connected direct, but need some logic levelers first.

Wiring is R1, G1, B1 to same on Hat.  R2, G2, B2, to same on Hat.  All 6 OE pins to one, both LE pins to one, and both STB pins to one. 

Thanks again.

hzeller, 2015-12-10:
Yes, for power supply make sure that you have good reserves. If you have the Pi on the same power supply (probably you do when you use the Adafruit HAT) make sure to have capacitors near to filter the spikes. You get 5V power supplies from eBay or you can make one out of an old PC power supply - they can do easily 25A.

The Adafruit HAT already has logic level convers, so you don't need extra converters after that (or did you mean that you started using the HAT because direct connect was not enough ?).

Wiring sounds good for now. if this actually has two sub-panels, then you don't need to uncomment DEFINES+=-DONLY_SINGLE_SUB_PANEL.

The mapping is probably just a long snake along the various rows, but you probably have to play around setting one pixel at a time to get an idea, then implement the transformer.

If you have a spare panel, I am happy to play with it, but I can't promise anything as this is a pure (overcommited-) spare time effort. eMail me directly if you need a snail-mail address.

closed 2017-07-30 by hzeller

anandrajgupta, 2021-11-16:
can someone please have a look at my issue of static scanning panel: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1329

I am not able to run it properly, infact the individual panel somewhat runs but chaining is not working at all, please help...