hzeller/rpi-rgb-led-matrix #105: "led-image-viewer"　new option

Kazu0015, 2015-12-12:
Hello, Hzeller.
I think "led-image-viewer" is very useful program. Thank you very much.
By the way, I wish two additional option.

The 1st is "the number of times" option.
if we execute "./led-image-viewer -n 3 test1.gif", A gif animation executes 3 times and terminate.
For example, I want to use like bellow.

./led-image-viewer -n 3 test1.gif
./led-image-viewer -n 2 test2.gif
./led-image-viewer -n 1 test3.gif

From the above, I want to execute multipule GIF animation continuously.

The 2nd is "time" option.
There is this option in "led-matrix" command.
 -t <seconds>  : Run for these number of seconds, then exit.
                        (if neither -d nor -t are supplied, waits for <RETURN>)

Please take it into consideration.

Regards,
kazu

ayman0201, 2015-12-19:
Hey Kazu,

I used http://gifmaker.me/ for that purpose.
First you put the images you want to make a gif with.
Then you specify the delay between each image and how many times you want them to repeat.

Hope that helps,
Ayman

hzeller, 2016-06-14:
Sounds like a good feature to have. Please send a pull request.

closed 2016-06-14 by hzeller